gnuradio/gnuradio #3865: QT GUI Sink display anomaly

duggabe, 2020-10-19:
When a QT GUI Sink is opened, the Frequency Display tab is selected, but the plot only occupies about 1/2 of the window width. As soon as the window is resized or another tab is selected, going back to the Frequency Display displays properly.

See ~/gnuradio/gr-qtgui/examples/gui_sink.grc

drmpeg, 2020-10-19:
Confirmed.

dl1ksv, 2020-11-03:
Works for me with the actual master branch

duggabe, 2020-11-03:
Ubuntu 20.04.1
GR/GRC 3.9.0.0-git built from source 26 Aug 2020
the first time you run it, if the window is very small, you might not see it. So make the window larger (maybe 1/2 screen), close that window, and then run it again. without touching it, the frequency plot will only occupy from the left side to underneath the 'Constellation Display' tab. then, resizing the window, or changing to another tab and going back will make the frequency plot full width.

duggabe, 2020-11-03:
[image]

dl1ksv, 2020-11-11:
Took some time to find the reason. But now it should be fixed in 3.9 by #3929

PR #3929 (2020-11-26): Resize the windows properly
There a some problems in resizing the windows.
Fixes #3865
closed 2020-11-26 by mormj