gnuradio/gnuradio #3871: 19 - qa_uncaught_exception (Failed), 197 - qa_header_payload_demux (Failed)

KyyB, 2020-10-20:
Hi Team,

Below is an error when running make test: 

The following tests FAILED:
	 19 - qa_uncaught_exception (Failed)
	197 - qa_header_payload_demux (Failed)
Errors while running CTest
Makefile:85: recipe for target 'test' failed
make: *** [test] Error 8

I resolved the other errors but was unable to resolve this.

Appreciate the help!

drmpeg, 2020-10-20:
These failures are known issues (caused by pybind11). There is no way to resolve them without fixing the root cause.

Currently, they are being ignored by the CI (Continuous Integration) builds. See:

https://github.com/gnuradio/gnuradio-buildbot/blob/public/config_module/builders/builders.json#L124

KyyB, 2020-10-21:
is it required to run GNU Radio? or is it possible without it?

drmpeg, 2020-10-21:
Failing qa_uncaught_exception doesn't have much effect. It only comes into play when there's an error.

Failing qa_header_payload_demux may have an effect if you're using the OFDM receiver in gr-digital.

closed 2020-10-27 by KyyB