gnuradio/gnuradio #3876: grc overwrites old (generated) file even after duplication and renaming

vu3rdd, 2020-10-20:
Seen in: GRC version: 3.8.2.0

To reproduce:

1. Create a flow graph in grc.
2. Click on File -> Duplicate.
3. Save the file with a new name.
4. Now, try generating the python file (Run -> Generate). At this point, GRC still writes to the old filename.py.

This is because, grc file's `options.parameters.id` field still points to the old filename and is not updated when it is saved.

Trivially worked around by opening the grc file with a text editor and changing the `id` value.

skoslowski, 2020-10-20:
This behavior is intended. The name of the generated module (for Python output) is the `id` set in the "Options" block - not the filename of the flow graph `example.grc`. Even if you duplicate a flowgraph, you may still want the same target name. Else, you can change the `id` parameter in the GUI - no need to fiddle with the saved file.

vu3rdd, 2020-10-21:
Thanks. As a user, I found the behaviour very confusing and was expecting `filename.grc` to write to <filename>.py, instead it kept writing to the old name. I didn't know that the `options` block came from the GUI (`options.parameters.id`).

Closing the issue. Thanks.

closed 2020-10-21 by vu3rdd