gnuradio/gnuradio #3877: Frequency Xlating FIR Filter block is not phase-continuous

daniestevez, 2020-10-21:
The Frequency Xltating FIR Filter block is not phase-continuous when the frequency is updated. The prototypical use case where this causes problems is when it's used to correct the the Doppler shift of a PSK signal.

The cause is that [`freq_xlating_fir_filter_impl::build_composite_fir()`](https://github.com/gnuradio/gnuradio/blob/912d48596569812ddcd05539bc7086617971b65e/gr-filter/lib/freq_xlating_fir_filter_impl.cc#L65`) doesn't take into account the current phase of the rotator `d_r` when rebuilding the taps of the FIR filter after a frequency change. I think that it can be fixed by multiplyin the taps by a suitable complex exponential that depends on the phase of `d_r` (I didn't think of the details, but I think it's easy to write the formula).

The [`rotator`](https://github.com/gnuradio/gnuradio/blob/912d48596569812ddcd05539bc7086617971b65e/gr-blocks/include/gnuradio/blocks/rotator.h#L22) doesn't have a method to get its current phase. We could do `d_r.rotate(gr_complex(1.0, 0.0))`, and then `d_r.set_phase()` with what we get (since calling `rotate()` increases the phase, which we don't want to do here). However, it seems much cleaner to add a `get_phase()` method to the rotator.

I'll probably send a PR with this in a few days/weeks, unless someone says here that they want to do it themselves.

willcode, 2020-10-28:
@daniestevez Is this possible? Even though we know the phase of the rotator, the cumulative phase shift due to the spun taps is unknown. So, we don't know what the effect of respinning the taps would be and can't correct for it. Am I missing something?

willcode, 2020-10-28:
Since the passband changes when the taps are respun, there's not even a guarantee that the same signals are being processed by the filter.

marcusmueller, 2020-10-29:
This sounds like a good idea to get right. However, why modify the phase of the filter if we could as well modify phase of the rotator?

willcode, 2020-10-29:
@marcusmueller That would be the way to do it ... but the change in phase during retune depends on the phase response of the filter. Even if the phase response is linear, it is unknown. I wasn't able to work out the math to make this work. Could just be my math :-)

daniestevez, 2020-10-29:
Hi! Yes, the idea to change the phase of the filter was sloppy on my part. It's much better to change the phase of the rotator. I bet we can derive a formula for this. I'll try to work out the math or some example.

In principle we have all the data we need here. We have the current rotator's phase, the previous FIR filter taps, and the new FIR filter taps. The correction should depend only on these parameters. However, it might happen that the formula we get is pretty ugly.

marcusmueller, 2020-10-29:
I think I was wrong:

So let's assume we're passing through a single eigenfunction $g=\exp(2i f t)$ at frequency $f$ through the unshifted linear system $h_0$. 

Then that experiences phase $\angle H_0(f)$, right? 

After shifting the system in frequency by $\Delta_f$, it experiences $\angle H_0(f-\Delta f)$; thus, the rotator needs to be offset by $-\angle H_0(f-\Delta f)$, far as I can tell. Small problem: 

That needs to be true for any $f$, but a rotator has but a single phase.

marcusmueller, 2020-10-29:
I only now realize that this thing doesn't do MathJax. Gimme a second.

marcusmueller, 2020-10-29:
Yeah, this feels true on multiple levels; can't have a different amplitude response while preserving the same phase response without reworking the whole filter!

I think in general, you wouldn't want that (you'd want the relative phase to still be the same, i.e. "5 Hz left of the center, the phase needs to be such and such relative to the center), and that's what shifting the whole filter achieves.

But still, what's the most appropriate phase you want give the rotator that moment? Well, that probably depends, but it feels like it should be such that what gets shifted to 0 Hz after the changeover keeps the same phase as it had before. 

@daniestevez does that describe a sensible analysis of the requirements of the doppler correction use case?

daniestevez, 2020-10-29:
I'll need to think more carefully about this (and try to see if I come up with a formula), but in my mind there is no obstruction.

Think of a multiply conjugate with a complex exponential followed by a decimating FIR filter. This system is clearly phase continuous on frequency updates as long as the complex exponential's frequency is update in a phase-continuous manner. Phase continuity of a sinusoid at the output doesn't depend on its input frequency.

Now, a frequency Xltaing FIR filter is just a way to rewrite this system to reduce the number of multiplications by the exponential. We push the exponential through the decimation using something like the noble identity. The prize we have to pay for this is to multiply the FIR taps by the exponential, so they become complex-valued.

So even without going into the mathematical details, I can't think of a reason why it shouldn't be possible to arrange the phase of the rotator in the frequency Xltaing FIR filter in such a way that phase continuity at the output is preserved, in the same sense as for the multiply conjugate + decimating FIR. But as said, until I'm able to work out a formula, I won't be 100% sure about this.

willcode, 2020-11-01:
Just in case the trivial solution would work, I did a quick experiment, adding std::conj(current rotator phase) to the taps. Got phase discontinuities on retune. Tried adding non-conjugated phase, same. I'll wait for better math!

daniestevez, 2020-11-02:
I've made [a small Jupyter notebook](https://github.com/daniestevez/jupyter_notebooks/blob/c84756642ca6ff0abb86ede23d3674955f4d7ede/Frequency%20Xlating%20phase%20continuity.ipynb) with a simulation to see if this is even possible, since @marcusmueller had concerns that the correction needed to preserve phase continuity might depend on the input frequency.

Honestly, I've used some guesswork to derive the appropriate correction for the phase rotator, but using this it seems to work correctly in all the cases I've tested. Perhaps it's good to grab a copy of the notebook and play around a bit more, trying to break it.

willcode, 2020-11-02:
Does that approximation only work for allpass filters? I plugged into gr_filter: FIR/LPF/Hamming, passband 0.25, stopband 0.30, 60dB. This gives 55 taps and -42 radians at f=0.25.

So, I think it depends on the taps, and something would need to determine the slope. Also, the linear slope only holds within the passband.

daniestevez, 2020-11-02:
The filters I used to test are (short) lowpass filters. I've just replaced with your suggestion, as generated by `gnuradio.filter.firdes.low_pass_2(1, 1, 0.25, 0.05, 60)` and it also works.

I've been doing some more tests, and I don't think the filter taps matter much, as long as the filter is something reasonable. For example, if I take the lowpass FIR taps and change more than half of the coefficients (the leftmost) by zero, then it doesn't work.

willcode, 2020-11-02:
OK, I'll put this in the GR code and see what happens.

willcode, 2020-11-03:
@daniestevez give this a try: https://github.com/willcode/gnuradio/tree/issue_3877

It seems to work fairly well in a few quick tests.

willcode, 2020-11-04:
GRC to experiment with the diff: https://gist.github.com/willcode/f32e905bc268eeebd0cf4cb95bb4b4f1

I think for the intended use, it works.

@argilo what do you think of this. Does it help anything where gpredict is command gqrx to do updates for tracking? It looks like you changed that to use xlate vs hw return earlier this year.

argilo, 2020-11-04:
The switch to Frequency Xlating FIR Filter in gqrx was purely for performance reasons. Originally it used Signal Source + Multiply, then briefly switched to Rotator before ending with Frequency Xlating FIR Filter. I don't personally use gqrx for anything where phase continuity would matter, but I suppose it would be a nice property to have.

willcode, 2020-11-04:
@daniestevez (N-1)/2 is the phase delay for symmetric taps, so it matches up with theory. I tried a few more cases, various LPF and a RRC. I'll clean up the naming a little more and put in a PR. I doubt anyone depends on the current behavior of arbitrary phase jumps.

daniestevez, 2020-11-04:
Thanks Jeff for all the tests!

Regarding GQRX, I wasn't aware that it is now using Frequency Xlating FIR Filter. People often do Doppler correction with GQRX, by sending frequency updates from gpredict. Back when it was using the the Signal Source + Multiply this worked well. I suspect that the current Frequency Xlating FIR might be causing issues to some people.

gamecat69, 2020-11-08:
This is really timely!

TLDR; I would really like to test the updated Xlating FIR filter with a real satellite signal for a Doppler correction solution I am working on. Would someone mind giving me some guidance on applying this update to my existing GNU Radio installation please? I am using v3.8 installed using PyBombs.

More info:
I have just implemented a Doppler shift solution which uses a traditional multiply + filter technique to offset the signal frequency. The offset is calculated using the relative velocity of the satellite and the ground station, which is calculated using the Two Line Elements (TLE) of the satellite. However, in some cases the TLE is not perfect (e.g. a Cubesat that drifts frequently in its orbital path) and I would really like to see how the new Xlating FIR filter works with the signals I am using.

daniestevez, 2020-11-08:
@gamecat69 the "Signal cosine source" + "Multiply" + "Filter" solution is already phase continuous. It is only the Frequency Xlating FIR filter the one that has the problem discussed here. So if you're using the Multiply, etc. I don't think anything here will help.

You're right that the TLEs are never perfect and there is always a residual after Doppler correction. In fact I spent some time looking at this for LEO small satellites and wrote a [series of posts](https://destevez.net/2017/08/wsjt-x-and-linear-satellites-part-i/). Hopefully, at 435 MHz the residual Doppler is only a few tens of Hz, and most noticeable for high elevation passes. Depending on your application, this may or may not be a problem. Also, depending on the satellite, thermal drift might be much worse than this effect. So you could consider estimating and tracking the residual frequency drift (FLL bandedge block, maybe?).

willcode, 2020-11-08:
@gamecat69 if you do decided it's useful, here's the patch with a couple things stripped out so it applies to maint-3.8. UNTESTED, and the QA code will fail for sure.

https://gist.github.com/willcode/1ecf84483a409585aae6c76e93effe73

Since this patch required a change to a .h file, I don't know whether it will find its way into 3.8 as is.

gamecat69, 2020-11-09:
Gents, thanks for the quick replies and the patch.

For context: The transmission I am working with is in the 2GHz range (S-band) and during the contact I see Doppler shifts from ~50Khz to + 50Khz.

@daniestevez: Thanks for the links to your post. It was both interesting and helpful. I actually used the same method to calculate the doppler correction frequency as your Python code. However, I previously used a rough value for C (300000000) and found your value provided more accurate results. I also found your observation that the doppler residual is highest when the satellite and ground station were closest - I used this observation to fine-tine my solution and now I have it working within reasonable limits. BTW: I am also part-way through your "Decoding Interplanetary Spacecraft" session on YouTube from GRCon Sept 2020 - nice work!

I may have misunderstood what the Xflating filter does.... I thought it may be able to apply a Doppler shift using a closed loop, rather than needing to receive an offset value from an external source - Is this correct or did I get it wrong? If correct, please would you explain how I would apply the patch and test this? Can I apply the patch with GNU radio already installed and compiled using PyBOMBS, or do I need to do a clean install and apply the patch at a particular point during the install?

willcode, 2020-11-09:
GR can not apply a Doppler shift unless the computer is moving really fast :-) The xlating filter combines a FIR filter (normally a LPF) and a frequency shift efficiently. It does not do anything you can't do in discrete steps.

You can patch a PyBOMBS install in the src/gnuradio directory and do a make install from src/gnuradio/build if you want. I do this for testing. It will probably mess things up if you do a pybombs update without cleaning up first. Or you could do a build from scratch. As Dani says, this might not be what you want, so it might not be worth the trouble. Or it could be fun.

daniestevez, 2020-11-09:
@gamecat69 The frequency Xlating filter is just a more efficient form of the "Complex cosine signal source -> Multiply conjugate -> Decimating lowpass filter", but mathematically it should be more or less equivalent (and in this issue we are discussing some implementation corner cases which don't make them exactly the same). It doesn't do any closed loop.

The main advantage of the frequency Xlating filter (besides having  less blocks in your flowgraph) is that it pushes the "multiply conjugate" operation after the decimation. So less multiplications are required. However, doing this requires a lowpass to bandpass transforms of the filter taps, which makes them complex, so you double the number of multiplications needed to implement the FIR. In practice I'm not sure what will be the performance advantage, since the implementation of the complex exponential in the "signal source" and the "Frequency Xlating FIR" is rather different. It will depend on the decimating factor and probably on whether your input signal is real or complex.

For a closed loop thing that tracks frequency drift (or Doppler), you might want to take a look at the FLL bandedge block.

willcode, 2020-11-25:
@daniestevez this has been merged. Test/close?

daniestevez, 2020-11-25:
Thanks! I'll see if I have some time to give it a try over the next week or so. If not, we can close this altogether.

willcode, 2020-12-17:
@daniestevez did you get time to check?

daniestevez, 2020-12-17:
Sorry! Not yet. Too much stuff got in the way.

Let's settle on something realistic: I try to push this an do some tests by tomorrow afternoon. If I don't manage, we just close this issue.

daniestevez, 2020-12-18:
I've finally given this a try, and I'm not sure it is working as expected. Maybe I'm not doing something correctly with my GNU Radio 3.9 install, since gnuradio-companion isn't able to compile .grc files to .py. I'm using a93d6c3bf2e20f3f3795fc631af33acc9ef1f187.

I'm using the Python script here for the test:
[test_xlating.zip](https://github.com/gnuradio/gnuradio/files/5718486/test_xlating.zip)
I'm also including the .grc file on which this Python script is based, but the Python script has additional by-hand modifications to change the centre frequency of the Frequency Xlating Filter a few times and plot the phase of the output with Matplotlib.

In the output plot I can see some jumps in the phase when the frequency changes, like this:

[image]

Incidentally, the output when running this under GNU Radio 3.8 looks pretty much the same.

Can you give this a try to check if there's something wrong with my GNU Radio 3.9 install?

willcode, 2020-12-19:
It's continuous when I run it on 3.9.
[image]
I do get the picture you show with the glitch under 3.8.

willcode, 2020-12-19:
@daniestevez The patch was added a month after https://github.com/gnuradio/gnuradio/commit/a93d6c3bf2e20f3f3795fc631af33acc9ef1f187.

daniestevez, 2020-12-19:
Now I see what I did! I git pulled from my Github fork, which is not updated with the upstream repo. I'll now pull from upstream and test again.

daniestevez, 2020-12-19:
I now tried it on a5e49d9120e8d0d7adf4de792cc3c4bc1a6defbc and it works perfectly. Even if you zoom all the way in to the frequency changes there is no phase glitch at all.

So I'm happily closing this as done. Thanks @willcode for all your help!

closed 2020-12-19 by daniestevez