gnuradio/gnuradio #3882: Polyphase Arbitrary Resampler functions differently on Raspberry Pi 4 and x86

guokr321, 2020-10-23:
Hello everybody,

Running environment:
X86 OS:
Linux ThinkPad-W541 5.4.0-47-generic #51~18.04.1-Ubuntu SMP Sat Sep 5 14:35:50 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux; DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.2 LTS"

Raspberry Pi 4 OS:
Linux raspberrypi 5.4.59-v7l+ #1338 SMP Fri Aug 28 11:55:37 BST 2020 armv7l GNU/Linux
Raspbian GNU/Linux 10 \n \l

gnuradio and uhd:
GRC 3.9.0.0-git and GRC-Maint3.7
UHD_4.0.0.0 and UHD_3.15.0

The attached grc can run correctly on X86, but on Raspberry Pi 4 it runs wrong.
[qpsk-test.zip](https://github.com/gnuradio/gnuradio/files/5426976/qpsk-test.zip)

 Would someone can help me？

dkozel, 2020-10-27:
Hi @guokr321 
Is GNU Radio 3.7 running on the Raspberry Pi? Which environment do you think is working correctly? Can you give some details about how the block is working differently?

guokr321, 2020-10-27:
Using GNU Radio main3.7 and 3.9.0.0 to test on on the Raspberry Pi 4 and PC repectively. PC can work correctly both, but on the Raspberry Pi 4 both wrong in received data.

bistromath, 2020-10-27:
You're going to have to be a lot more specific. How is it wrong? How are
you testing? Do you have an example flowgraph showing the problem? Can you
provide a plot of the results?

marcusmueller, 2020-10-28:
Hi @guokr321, thank you for reporting! Sadly, I have no Pi4 at my disposal to test myself.

Sadly, it's really not clear what you're testing here. Or really, what this flow graph does. Please, when sharing such flow graphs, make sure they are tidier:

[image]

We'll need to reduce that test case to something that eliminates other sources of error. In this case, you'd definitely would want to take all the USRP out of the problem.

Because, for example, wild guess: Your x86 computer is computationally much more powerful than your Raspberry Pi, and you're just seeing USRP/UHD over- or underflows.

We can reopen this case once there's conclusive indication it's actually the arbitrary resampler that is problematic on the raspberry pi, and not the overall application.

guokr321, 2020-10-29:
the testing are following:
    On the X86, I run the attached grc, using GNU Radio maint3.9 and 3.9.0.0, the constellion in the receiver is correct, and write the sender and receiver data into files, compare the files, and find the receiver data contains the whole sender data.
    On the  Raspberry Pi 4, do the same thing as X86, the receiver constellaion is disordered.
    Then, split the grc into the sender grc and receiver grc, using two USRP, one is sender, and another is receiver. Using X86 to be sender, Raspberry Pi 4 as receiver, the received data is correct; using Raspberry Pi 4 as sender, X86 as receiver, the received data are wrong.

marcusmueller, 2020-10-29:
Interesting, but that might have very very many different causes. How do you know it's the resampler?

willcode, 2020-10-29:
rrc_taps in your .grc has length 1409. At 1MS/s, that could be a bit much for a RPi.

guokr321, 2020-10-30:
But using Throttle and Chammel Model insteading of USRP, the received 'filebitout' file is consistent with input 'filebitin' file on Raspberry Pi 4.

guokr321, 2020-10-30:
Moreover, on Raspberry Pi 4 the CPU capacity has harf of the total, memory has only one twentieth.

marcusmueller, 2020-10-30:
> Moreover, on Raspberry Pi 4 the CPU capacity has half of the total, memory has only one twentieth.

That's not how that works, at all.

> filebitout

Yeah, so this is almost certainly not a bug. Your Pi is just slower than your PC, it's not keeping up with the USRP, and hence samples are dropped. You would be seeing `U`, `O`, `D` or `S` printed during operation.

closed 2020-10-30 by marcusmueller