gnuradio/gnuradio #3883: cpu_features build issue -> failure w/ pybombs install

cclamb, 2020-10-23:
Guys, not sure where to put this; also reported to the Volk crew. Anyway,  cpu-features in Volk is a submodule but when I install gnuradio via pybombs the git checkout doesn't execute recursively so the cpu-features submodule isn't checked out. As a result the pybombs build fails. To recreate:

pybombs auto-config
pybombs recipes add-defaults
pybombs prefix init ~/gnuradio -R gnuradio-default

Thanks!

ValZapod, 2020-10-24:
Well, you need to do git submodule init && git submodule update

cclamb, 2020-10-24:
Yep, and that does fix it. The problem is that pybombs doesn't do this. I can use the gitargs: --recursive option with pybombs too, and that will fix it.

The issue is that it should work by default - I shouldn't need to manually execute git commands during the pybombs build or use a special flag to get it to work.

Issues or related issues I've opened with the Volk crew.
https://github.com/gnuradio/gr-recipes/pull/238
https://github.com/gnuradio/volk/issues/410

dkozel, 2020-10-24:
Hi Chris. Thanks for the report. I'm going to close the one here as the issue on the PyBOMBS gr-recipes repo is the best place to track the progress of the fix.

Cheers, Derek

closed 2020-10-24 by dkozel