gnuradio/gnuradio #3884: Pybind: Module loading failures get masked in the init

mormj, 2020-10-23:
If there is some linker error, or another issue with the pybind in an OOT, this gets masked because of 

```
except ImportError:
    pass
```
in `__init__.py`

There are some situations where the ImportErrors should be caught and ignored, such as a python only module (which is why the pass is there).  But in other situations, the underlying error should be displayed

marcusmueller, 2020-11-07:
we really should get rid of this. When people port their modules to pybind/GR 3.9, we'll have loads of these errors, and the first time we'll always tell people is to remove these two lines.

marcusmueller, 2020-11-07:
Proposal: special handling for "can't find the module" vs any other error:

* Any other error: gets bubbled up
* failure to find module: log and pass

Question is whether we can rely on GR logging facilities in `__init__.py`.

PR #3926 (2020-12-17): modtool: Change handled exception to ModuleNotFoundError
Change the pass on exception to pass only on ModuleNotFoundError, not any ImportError

This should make the behavior as follows:
1) Python only module should import without error (by passing at ModuleNotFoundError)
2) C++ module with missing symbol should be caught and displayed

Tested 2) by creating a block with a dummy function in the header with no implementation.  Then call that implementation from another function.  Compiles ok, but will cause undefined symbol error at import.

Fixes #3884
closed 2020-12-17 by mormj