gnuradio/gnuradio #3888: Osmocom Blocks not Available

KyyB, 2020-10-26:
Hi everyone,

I cannot seem to find the Osmocom source & other blocks related to osmocom. Why is that?

willcode, 2020-10-27:
Please include GNU Radio version and other information. Are you trying to build/install the gr-osmosdr out-of-tree module?

KyyB, 2020-10-27:
GNU Radio Companion 3.8.2.0 and osmocom block and source block were not part of the package

dkozel, 2020-10-27:
Hi Kyyb. The Osmocom blocks aren't part of the main GNU Radio code. You need to install gr-osmosdr separately.
https://osmocom.org/projects/gr-osmosdr/wiki

I hope you're able to get it installed and running. The GNU Radio chat and mailing list are both good places to ask questions, and Osmocom have their own chat and mailing lists as well if you have questions about Osmocom blocks.

closed 2020-10-27 by dkozel