gnuradio/gnuradio #3889: Errors when running Make (building from source)

KyyB, 2020-10-27:
Hey everyone,

I am building GNURadio from the source & got following errors when running Make:

//usr/lib/x86_64-linux-gnu/libSM.so.6: undefined reference to `uuid_generate@UUID_1.0'
//usr/lib/x86_64-linux-gnu/libSM.so.6: undefined reference to `uuid_unparse_lower@UUID_1.0'
collect2: error: ld returned 1 exit status
gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/build.make:147: recipe for target 'gr-qtgui/examples/c++/display_qt' failed
make[2]: *** [gr-qtgui/examples/c++/display_qt] Error 1
CMakeFiles/Makefile2:7131: recipe for target 'gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/all' failed
make[1]: *** [gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/all] Error 2
Makefile:159: recipe for target 'all' failed
make: *** [all] Error 2

Appreciate the support!

KyyB, 2020-10-27:
I ran ls /usr/lib | grep qt
and got the below:

libqwt-qt5.so
libqwt-qt5.so.6
libqwt-qt5.so.6.1
libqwt-qt5.so.6.1.3
qt5

dkozel, 2020-10-27:
Hi KyyB. Can you please let us know what operating system (what version of Linux) you're using, and which branch of GNU Radio?

closed 2020-10-27 by KyyB