gnuradio/gnuradio #3891: GNURADIO is running but while executing/capturing i got errors

KyyB, 2020-10-27:
Hey Everyone 

I have faced issue while executing/capturing

Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 14, in swig_import_helper
    return importlib.import_module(mname)
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 994, in _gcd_import
  File "<frozen importlib._bootstrap>", line 971, in _find_and_load
  File "<frozen importlib._bootstrap>", line 955, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 658, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 571, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 922, in create_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
ImportError: /usr/local/lib/x86_64-linux-gnu/libgnuradio-osmosdr.so.0.2.0: undefined symbol: _ZN2gr10fcdproplus3fcd4makeENSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/testbed/gr-osmosdr/Test/top_block.py", line 34, in <module>
    import osmosdr
  File "/usr/local/lib/python3/dist-packages/osmosdr/__init__.py", line 25, in <module>
    from .osmosdr_swig import *
  File "/usr/local/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 17, in <module>
    _osmosdr_swig = swig_import_helper()
  File "/usr/local/lib/python3/dist-packages/osmosdr/osmosdr_swig.py", line 16, in swig_import_helper
    return importlib.import_module('_osmosdr_swig')
  File "/usr/lib/python3.6/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named '_osmosdr_swig'

Thanks & appreciate the help.

KyyB, 2020-10-27:
GNU Radio 3.8.2.0 (Python 3.6.9)

willcode, 2020-10-27:
Hi @KyyB . This is not a gnuradio issue. It has to do with gr-osmosdr, or more likely your environment. Please use chat/mail for help requests. I'm sure someone will be happy to walk through the setup with you.

closed 2020-10-27 by willcode

KyyB, 2020-10-27:
Where can I find the chat or mail?

willcode, 2020-10-27:
https://wiki.gnuradio.org/index.php/HowToGetInvolved