gnuradio/gnuradio #3893: Port install gnuradio fatal error on open

Hook12aaa, 2020-10-27:
Hi,

I typed the following command into terminal to install gnuradio : "sudo port install gnuradio".  To replicate this error use the current version of macos with the current version of macport.  Do the command and then type "gnuradio-companion ".

`raceback (most recent call last):
  File "/opt/local/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/opt/local/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/main.py", line 73, in main
    from .gui.Platform import Platform
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/Platform.py", line 26, in <module>
    from . import canvas
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/canvas/__init__.py", line 21, in <module>
    from .param import Param
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/canvas/param.py", line 23, in <module>
    from .. import ParamWidgets, Utils, Constants
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 69, in <module>
    add_style_provider()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 67, in add_style_provider
    Gtk.STYLE_PROVIDER_PRIORITY_APPLICATION
TypeError: Argument 0 does not allow None as a value`

marcusmueller, 2020-10-30:
could you add what `gnuradio-config-info --version` says about your machine?

Hook12aaa, 2020-11-02:
Sure @marcusmueller it returned '3.8.2.0'

nirmita123, 2020-11-18:
I got the same error on the same gnuradio version. @Hook12aaa were you able to solve the problem?

(gnuradio-companion:95507): Gtk-WARNING **: 20:27:04.948: Locale not supported by C library.
	Using the fallback 'C' locale.
Traceback (most recent call last):
  File "/opt/local/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/opt/local/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/main.py", line 73, in main
    from .gui.Platform import Platform
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/Platform.py", line 26, in <module>
    from . import canvas
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/canvas/__init__.py", line 21, in <module>
    from .param import Param
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/canvas/param.py", line 23, in <module>
    from .. import ParamWidgets, Utils, Constants
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 69, in <module>
    add_style_provider()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 67, in add_style_provider
    Gtk.STYLE_PROVIDER_PRIORITY_APPLICATION
TypeError: Argument 0 does not allow None as a value

Hook12aaa, 2020-11-18:
Nope and no one responded to my thread. I haven't been able to use the program at all

alvarolopez12, 2020-11-18:
Any update for this??

nirmita123, 2020-11-18:
I followed the link mentioned below and it worked fine for me.  https://aaronscher.com/wireless_com_SDR/MacOSX_install_gnu_radio.html

rgov, 2020-12-03:
The Homebrew formula works. I've been testing with it lately. There is a known crash in Cairo, a dependency, on Big Sur, but a fix is forthcoming.

mh-, 2020-12-27:
`Gdk.Screen.get_default()` returns `none`

gefa, 2021-01-06:
@Hook12aaa I can reproduce this exact error. However, if I launch gnuradio-companion from: Applications folder -> MacPorts -> GNURadio, then it works fine.

nickoe, 2021-01-14:
This seems to be a packaging issue and not gnuradio itself. Possibly the thing in whatever is mentioned in https://github.com/gnuradio/gnuradio/issues/3893#issuecomment-755023784 sets some environment variables which are not set in the system shell.

jImbyers, 2021-01-30:
Getting these error when trying to use GNU Companion from terminal
Launching from MacPorts folder does not work either

MacOS 11.1
GNU version 3.8.2.0
Installations via MacPorts 2.6.4

Any help will be appreciated 

73
Jim 
VE7KAJ  VE3AJ

 ~ % gnuradio-companion
dbus[28454]: Dynamic session lookup supported but failed: launchd did not provide a socket path, verify that org.freedesktop.dbus-session.plist is loaded!
No such schema “org.gnome.desktop.interface”
Traceback (most recent call last):
  File "/opt/local/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/opt/local/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/main.py", line 83, in main
    platform.build_library()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/core/platform.py", line 197, in build_library
    utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/collections/__init__.py", line 916, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/collections/__init__.py", line 908, in __missing__
    raise KeyError(key)
KeyError: 'options'

sundial-dreams, 2021-04-30:
➜  ~ gnuradio-companion
No such schema “org.gnome.desktop.interface”
Traceback (most recent call last):
  File "/opt/local/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/opt/local/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/main.py", line 73, in main
    from .gui.Platform import Platform
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/Platform.py", line 26, in <module>
    from . import canvas
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/canvas/__init__.py", line 21, in <module>
    from .param import Param
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/canvas/param.py", line 23, in <module>
    from .. import ParamWidgets, Utils, Constants
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 69, in <module>
    add_style_provider()
  File "/opt/local/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/gnuradio/grc/gui/ParamWidgets.py", line 67, in add_style_provider
    Gtk.STYLE_PROVIDER_PRIORITY_APPLICATION
TypeError: Argument 0 does not allow None as a value

rixon, 2021-08-17:
I have had the same error, and have found what caused it in my case.  Note the error starts with "No such schema..." and ends with the "TypeError" as shown in the previous comment.

I tried launching from the command line in iTerm2.  I suspected an X problem.  XQuartz was not running.  Next I ran xeyes and it failed because there was no display.  I then started XQuartx, tested xeyes again, which worked, so then started gnuradio-companion again - this time it worked.

**In short, the problem was because there was no X server running, and XQuartz needed to be started manually.**

My environment, for reference:
macOS Big Sur 11.5.1
XQuartz 2.8.1
iTerm2 3.4.8
Port 2.7.1

michaelld, 2021-09-25:
@rixon Great summary! I've moved to using default variants with MacPorts of `-x11 +no_x11 +quartz` and so I don't need X11 for running GRC. Works nicely! For folks using MacPorts who just want it for GNU Radio, I highly recommend using these variants by default. I do -not- recommend setting `+python38` nor `+gcc11` or the equivalent; those generally work, but make it very difficult to update to newer / different Python or GCC.

I'm closing this issue since it's not directly a GR one, and there is good info on how to deal with it. If anyone objects, please reopen and post relevant info.

closed 2021-09-25 by michaelld

wlopez-enkoding, 2022-05-10:
> @rixon Great summary! I've moved to using default variants with MacPorts of `-x11 +no_x11 +quartz` and so I don't need X11 for running GRC. Works nicely! For folks using MacPorts who just want it [...]

Is this still the way to go ?

I'm getting `Failed to configure gnuradio: gtk3 must be installed with +quartz`