gnuradio/gnuradio #3911: GNU Radio 3.8.2 no longer works built with Python2

drmpeg, 2020-11-01:
Seems to have been broken with this commit:

https://github.com/gnuradio/gnuradio/commit/507e85722573cfdc3029b4fd7776a75987f332d7#diff-74872b8e57a17b66548d67e413291f575b4d2aeca8f382c8834d89c0719cf282

Fails with the following.
```
re@beavis:~/xfer$ gnuradio-companion sinetest.grc
Traceback (most recent call last):
  File "/opt/gnuradio-3.9.0git/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/opt/gnuradio-3.9.0git/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/opt/gnuradio-3.9.0git/lib/python2.7/dist-packages/gnuradio/grc/main.py", line 73, in main
    from .gui.Platform import Platform
  File "/opt/gnuradio-3.9.0git/lib/python2.7/dist-packages/gnuradio/grc/gui/Platform.py", line 26, in <module>
    from . import canvas
  File "/opt/gnuradio-3.9.0git/lib/python2.7/dist-packages/gnuradio/grc/gui/canvas/__init__.py", line 21, in <module>
    from .param import Param
  File "/opt/gnuradio-3.9.0git/lib/python2.7/dist-packages/gnuradio/grc/gui/canvas/param.py", line 23, in <module>
    from .. import ParamWidgets, Utils, Constants
  File "/opt/gnuradio-3.9.0git/lib/python2.7/dist-packages/gnuradio/grc/gui/ParamWidgets.py", line 20, in <module>
    import configparser
ImportError: No module named configparser
re@beavis:~/xfer$
```

drmpeg, 2020-11-01:
Ignore the /opt/gnuradio-3.9.0git/ install directory name. I just use that over and over.

ewxl, 2020-11-27:
Not that it matters, but it broke for me last year or something like that, when gr_threading was patched with code that did not consider python2.6. Anyway, since python2 is end of life, I'm not sure this will be adressed?

drmpeg, 2020-12-01:
The only reason I filed this issue is for Windows builds. The Windows build here:

http://www.gcndevelopment.com/gnuradio/downloads.htm

depends on Python 2.7. Notice that he hasn't done a 3.8.2 build.

I guess there's a new way to do Windows installs with Anaconda, but I don't know much about it.

nickoe, 2021-01-10:
@drmpeg Are you sure this is a real problem? I have the `configparser` module available for me in python2.
```
Python 2.7.18 (default, Sep  5 2020, 11:17:26) 
[GCC 10.2.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import configparser
>>>
```
No errors when importing.

drmpeg, 2021-01-11:
Thanks @nickoe I overlooked the obvious on this. Installing ```python-configparser``` solves the issue. However, it would seem that ```python-configparser``` is an additional dependency since it was not on either of my Ubuntu 18.04 systems.

closed 2021-01-11 by drmpeg

nickoe, 2021-01-11:
By the way, the commit @drmpeg  refers to in the description is not on 3.8, but the same commit is cherry picked to maint-3.8 as 8c6872594f10eb06f694a89afb407af2624787d8.