gnuradio/gnuradio #3914: QT GUI Time Sink changing parameters in multiple tabs don't get saved

duggabe, 2020-11-03:
Ubuntu 20.04.1
GR/GRC 3.9.0.0-git built from source 26 Aug 2020
test1103.grc
- Signal Source => Throttle => QT GUI Time Sink
-  execute without changes
-  close window
-  open QT GUI Time Sink Properties
-  General tab
-  change Number of Points to 256
-  Apply
-  Config tab
-  Control Panel - Yes
-  Apply
-  OK
-  execute
-  the Number of Points is changed, but there is no Control Panel

I have noticed this on several of the QT GUI sinks, but thought maybe I was doing something wrong. This is a repeatable procedure. The first change gets saved; changes in other tabs do not get saved.

Here is a snippet of the Python code:
```
        ##################################################
        # Blocks
        ##################################################
        self.qtgui_time_sink_x_0 = qtgui.time_sink_c(
            256, #size
            samp_rate, #samp_rate
            "", #name
            1, #number of inputs
            None # parent
        )
        self.qtgui_time_sink_x_0.set_update_time(0.10)
        self.qtgui_time_sink_x_0.set_y_axis(-1, 1)

        self.qtgui_time_sink_x_0.set_y_label('Amplitude', "")

        self.qtgui_time_sink_x_0.enable_tags(True)
        self.qtgui_time_sink_x_0.set_trigger_mode(qtgui.TRIG_MODE_FREE, qtgui.TRIG_SLOPE_POS, 0.0, 0, 0, "")
        self.qtgui_time_sink_x_0.enable_autoscale(False)
        self.qtgui_time_sink_x_0.enable_grid(False)
        self.qtgui_time_sink_x_0.enable_axis_labels(True)
        self.qtgui_time_sink_x_0.enable_control_panel(False)
        self.qtgui_time_sink_x_0.enable_stem_plot(False)
```

dave-pi, 2020-11-04:
I've reproduced in 3.8.2.0 Also I think it's more to do with multiple 'Applies'. Example: Change points from 256->512, hit Apply, change points from 512->1024, hit Apply. Second Apply doesn't get saved.

dave-pi, 2020-11-05:
The issue was introduced in this commit: b2d4a832823d097058a8e8d549ce355c2d3cf573. The problem is that `flow_graph.import_data(n)` creates complete new blocks in the flowgraph, and so the block that was passed to the dialog box for editing, is no longer the block being updated. In other words, after the first 'Apply' you can 'Apply' again all you want but the  the block stored in the dialog box is no longer the block in the flowgraph because the later was replaced in the offending line.
If I comment out just that line, and leave the rest of the changes, everything works fine. Both the original issue and this one are gone. @dl1ksv if you could pitch in, wrt to the need for the import_data line would be great to make sure I'm not breaking anything else.

dl1ksv, 2020-11-06:
That was in the early days of 3.8. It was a bit tricky, I have to take a deeper look.

dl1ksv, 2020-11-11:
@dave-pi I  removed https://github.com/gnuradio/gnuradio/blob/55621a9709b219551b908e67ee88f6f7ad2593cb/grc/gui/Application.py#L549 and it worked.

dave-pi, 2020-11-11:
OK, I'll move ahead with a PR and do some more testing just to make sure. Tks.

PR #3932 (2020-12-17): grc: Fix multiple 'Apply' in modify block parameters dialog box
When the modify parameters dialog block is constructed, the instance of
the block to be modified, is passed in. When the apply button is clicked
there is an extraenous flowgraph.import_data() call that re-creates all
blocks in the flowgraph. The consequence is that the block instance
passed to the dialog box is no longer the one in the flowgraph, and so
subsequent Applies have no effect. This change simply removes this call.

Fixes: #3914
closed 2020-12-17 by mormj