gnuradio/gnuradio #3918: QT GUI Chooser causing problems in GNU Radio v3.8.2

bobbryce, 2020-11-04:
Adding the QT GUI Chooser block to a simple working scheme causes a failure to RUN and generates error code.

Environment is new GNU Radio v3.8.2 installation on Linux Mint 20 with Python 3.8.5.
Working schematic is a Signal Source into a QT GUI Sink via a Throttle. A QT GUI Range slider is used to control the frequency - all works fine.

Simple addition of adding the (unconfigured) Chooser to the schematic causes the issue. Correctly configuring the Chooser for the intended operation does not remove the problem.

The error starts with -  Generate Error: (NameError("'i' is not defined")......

Two files attached - screen snapshot of the schematic and a text file of the error.

[image]

[Simple_project_plus_configured_GUI_Chooser.txt](https://github.com/gnuradio/gnuradio/files/5490144/Simple_project_plus_configured_GUI_Chooser.txt)

drmpeg, 2020-11-04:
I can't reproduce this. Are you sure you're actually using GNU Radio 3.8.2? It seems this bug was fixed for 3.8.2.

https://github.com/gnuradio/gnuradio/commit/6964b6b6839ef5db63fbfacdd721ae58a75c70a8

bobbryce, 2020-11-04:
Yes.
Confirmed by using Help/About - v3.8.2.0
Installed through typing
sudo add-apt-repository ppa:gnuradio/gnuradio-releases
etc

drmpeg, 2020-11-05:
Something's not right with you installation. Notice that your Signal Source block only has a frequency message input. But a command message input was added for GNU Radio 3.8.2. Like so:

[image]

bobbryce, 2020-11-05:
Strange!!!!
Did you notice that your welcome message says 3.8.2.0 but your Block path says 3.9.0 -  so, what version are you using?

A copy of my screen attached - no Block reference to 3.9.0.

I guess we need to rule out the version issue but also whether or not there are differences caused by being on Linux Mint 20?

I first installed via the Linux Mint Software Manager menu item and ended up with what appeared to be version 3.8.1.
This was then removed and I installed via the following commands - tis gave me what appears to be v3.8.2.0
sudo add-apt-repository ppa:gnuradio/gnuradio-releases
sudo apt-get update
sudo apt-get install gnuradio

PS Appreciate your help in trying to fix this.

[image]

drmpeg, 2020-11-05:
My block path is a result of building 3.8.2 from source and re-using the same install directory from previous builds. I know it's confusing, but it doesn't have anything to do with 3.9.

Your problem is there's still remnants of 3.8.1 in your install. Another tell-tale is your QT GUI chooser block. Notice it doesn't have a separate ```Default option``` like mine does. That's another change that went into 3.8.2.

I'm not exactly sure how to fix it. You could try ```sudo apt-get purge gnuradio``` and then install again. If that doesn't work, then you may have to manually delete GNU Radio files in /usr/lib, /usr/bin, /usr/share and /usr/include. But be extremely careful with that, since that's where all of your other installed programs reside. If it's a new system, then a fresh install of Mint may be the easiest way to go.

bobbryce, 2020-11-05:
Wow!
Thanks very much for the assessment - although it's starting to sound scary to a new Linux user!
Are we ruling out the version of Linux as a possible confusion - I know (as a Windows 10 user) I'm still confused about the methods used to install and delete applications in Linux. Does my install method sound OK to you, or do you recommend the fresh build approach - I've never tried that before, but we only live once eh?

I'll start first with another attempt at deleting and installing GNU Radio and will look for the tell-tale differences in the new install to check the actual version I have achieved.

Thanks for the assistance.

bobbryce, 2020-11-05:
Summary - deleted the gnuradio installation, checked, re-installed - same result - see attached screenshot.
[image]

What I did:-
* Used the Software Sources utility in Linux Mint to remove the PPA. There were two but only the v3.8.2 was enabled - both were removed.
* sudo apt-get purge gnuradio
* sudo apt-get autoclean
* checked the suggested folders for remnants (there were none) - usr/lib, /share and /include
* re-installed using sudo add-apt-repository ppa:gnuradio/gnuradio-releases, then sudo apt-get update, then sudo apt-get install gnuradio

As can be seen from the attached screenshot, the Frequency Source still does not show the 'cmd' option and the layout of the QT GUI Chooser is different from yours.

Next option is apparently to trash the Linux installation. I'm reluctant to do that as there is a fair bit of time invested in that.

duggabe, 2020-11-05:
FYI: look at the note in https://wiki.gnuradio.org/index.php/QT_GUI_Chooser about the change in parameters. Depending on when your package was created, you could have one or the other of the selections.

I'm curious: how did you choose '101' and '103' as the selections of the waveforms?

bobbryce, 2020-11-05:
1) Even stranger having looked at this reference. Given what appears in my system (see screenshot), I have a version of the Chooser that precedes version 3.8.1, since it shows the 'old' version. This is despite never having installed a pre v3.8.1 version and despite both GNU Radio and Help/About declaring it version 3.8.2.0.
[image]

2) Tooltips - If the cursor is allowed to hover over the Signal Source/Waveform setting, it shows a value. With the default setting of 'Cosine', it shows a value of 102, so the values I used are seen by setting the different options and using the values from the Tooltip.

Would building the application from sources get around this problem do you think?

Many thanks for your help in this - I'm still keen to use GNU Radio, and have a hardware purchase on hold and depending on whether I can get this app to work or not.....

bobbryce, 2020-11-05:
Sadly, my machine is now croc.
In deleting v3.8.1 files in /usr/lib, /share and /include, something critical has been deleted, so no more Linux machine.
I thought Timestamp was supposed to save me.....

Thanks for trying to help. I may come back to this thread at some point.

bobbryce, 2020-11-05:
OK! System restored! Always a first time for everything!

So, my question is now - research suggests that there *may* be a problem with MINT 20 installing PPAs, but when I use the recommended install command (sudo apt install gnuradio) I keep getting v 3.8.1 rc2.
Can you please recommend how to go about installing the latest version of GNU Radio on Linux Mint 20?

Many thanks.

dave-pi, 2020-11-05:
You might be better off installing from source: https://wiki.gnuradio.org/index.php/InstallingGR#For_GNU_Radio_3.8_or_Earlier

bobbryce, 2020-11-05:
dave-pi - Thanks for comments. That looks well scary for a Linux newbie, but at least I think I have mastered backups now so I might try it.

The point of this update is to report that I restored a Mint 20 backup from a restore three weeks ago before I even thought of GNU Radio. After this restore, I checked that there was no evidence of GNU Radio either through system managers or through searching through files.

I proved that my repositories were only able to deliver GNU Radio versions 3.8.1, and so I went with the PPA install method. This has delivered v 3.8.2.0 as confirmed by watching the installs, by checking the Help/About and by manually inspecting the files in /usr/lib and elsewhere.

I am now as convinced as I can be that I have downloaded and correctly installed v3.8.2.0.

I still get the same errors - see attached, and am still seeing blocks that I've been told are the wrong version blocks for v3.8.2.0.   Something is clearly wrong here.

If going through a build from source is the only way then I'll give it a go, but I'm apprehensive!

[Chooser error.txt](https://github.com/gnuradio/gnuradio/files/5498042/Chooser.error.txt)
[image]

dave-pi, 2020-11-06:
- what is the location of gnuradio-companion (you can type `which gnuradio-companion`)?
- what are the block locations reported by gnuradio-companion when it starts up?

bobbryce, 2020-11-06:
Dave-Pi
Thanks for the query - the answers are:-
location: 
/usr/bin/gnuradio-companion

Block paths:
	/usr/share/gnuradio/grc/blocks

dave-pi, 2020-11-06:
Try downloading the .deb file, uncompressing it and checking that particular Signal Source block file to look for the missing 'cmd' message.
```
sudo apt-get download gnuradio
dpkg -x gnuradio_3.8.2.0-0~gnuradio~bionic-1_amd64.deb tmp
grep cmd tmp/usr/share/gnuradio/grc/blocks/*
```

bobbryce, 2020-11-06:
David-Pi
Tried what you requested and got two errors - one was a permission error, the other related to your use of 'bionic-1'. I modified the command to one that seemed to fit and got the response below. Hope that helps?

bob@SamsungLaptop:~$ sudo apt-get download gnuradio
[sudo] password for bob:     
Get:1 http://ppa.launchpad.net/gnuradio/gnuradio-releases/ubuntu focal/main amd64 gnuradio amd64 3.8.2.0-0~gnuradio~focal-2 [162 MB]
Fetched 162 MB in 34s (4,825 kB/s)                                                              
W: Download is performed unsandboxed as root, as file '/home/bob/gnuradio_3.8.2.0-0~gnuradio~focal-2_amd64.deb' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)
bob@SamsungLaptop:~$ dpkg -x gnuradio_3.8.2.0-0~gnuradio~bionic-1_amd64.deb tmp
dpkg-deb: error: failed to read archive 'gnuradio_3.8.2.0-0~gnuradio~bionic-1_amd64.deb': No such file or directory
bob@SamsungLaptop:~$ dpkg -x gnuradio_3.8.2.0-0~gnuradio~focal-2_amd64.deb tmp
bob@SamsungLaptop:~$ grep cmd tmp/usr/share/gnuradio/grc/blocks/*
tmp/usr/share/gnuradio/grc/blocks/analog_sig_source_x.block.yml:    id: cmd
bob@SamsungLaptop:~$

bobbryce, 2020-11-06:
PS Not sure why parts of this copy and paste has random strike-throughs?
Re-sent the response as a text file attached.
[grep.txt](https://github.com/gnuradio/gnuradio/files/5501222/grep.txt)

dave-pi, 2020-11-06:
I would now run the same grep command on the installed block to see if it's the right one:
`grep cmd /usr/share/gnuradio/grc/blocks/*`

if it's correct then there is a problem in gnuradio-companion finding the blocks.

if not, install Gnuradio from the .deb:

First double check where it's going to install that Sig Source file:
`dpkg -c gnuradio_3.8.2.0-0~gnuradio~bionic-1_amd64.deb | grep analog_sig_source`

then install:
`sudo dpkg -i gnuradio_3.8.2.0-0~gnuradio~bionic-1_amd64.deb`

then run the grep again to see:
`grep cmd /usr/share/gnuradio/grc/blocks/*`

bobbryce, 2020-11-06:
Hi dave-pi
The installed version appears to be the same as that in the .deb - see the attached text.
[grep_installed.txt](https://github.com/gnuradio/gnuradio/files/5501547/grep_installed.txt)

I also include a listing of the blocks folder, but that may not be helpful?
[blocks folder listing.txt](https://github.com/gnuradio/gnuradio/files/5501579/blocks.folder.listing.txt)

I did not proceed to install again as there seemed no point given the above.

I'm thinking of starting again and installing the latest version from here, or else going back to an earlier version like 3.7....
https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-master
Any thoughts?

dave-pi, 2020-11-06:
It's strange, your installation seem sane.
Try this, delete the .yml for the signal source block(not really delete it, just move it somewhere else) and start gnuradio-companion to see if it shows up in the block list.
Or in alternative save a flowgraph with that block, close it, remove the .yml, and then open it again to see if it detects the missing block.

bobbryce, 2020-11-06:
Renamed the .yml to .txt

[image]

bobbryce, 2020-11-06:
/usr/share/gnuradio/grc/blocks/analog_sig_source_x.txt
[analog_sig_source_x.txt](https://github.com/gnuradio/gnuradio/files/5502632/analog_sig_source_x.txt)

dave-pi, 2020-11-06:
looks good. Rename back. Remove the Gui Range and Chooser and run the flowgrpah with just the signal source, the throttle and the sink.

bobbryce, 2020-11-06:
Hi dave-pi,
Thanks again for trying to assist.
Unless I misunderstand, what you're asking gives no new information - GNU Radio runs fine in this configuration - it was adding the Chooser that caused the errors. You since added your doubts on the version number by noting that the Signal Source did not look right and should have the CMD input - I've never seen that on v3.8.2.0, and as has been checked several times, I only have the code for 3.8.2.0 (unless the packaging of this PPA was incorrect).

See attached flow diagram that has the Signal Source operating but without a Chooser block.
[image]

If we are fresh out of ideas, then I'm tempted to try for the early v3.9. What are your throughts?

Thanks again

dave-pi, 2020-11-06:
positively bizarre! anyway I really can't think of anything else right now short of debugging gnuradio-companion to check what's happening.
As a last desperate attempt I would just remove grc preferences ~/.gnuradio/grc.conf, but it's mostly wishful thinking.
If you go for source build just stick with 3.8, because 3.7 is ancient and 3.9 is a moving target. Good luck!

bobbryce, 2020-11-06:
Ok dave-pi. 
Removing the conf file did not help. As you would expect, it started me with a clean slate. However other than that, there was no difference in behaviours.

Thanks for the efforts anyway. 
Have a great weekend!

bobbryce, 2020-11-07:
In Conclusion.

As a final effort to get to a working GNU Radio, I took my system back to a clean state and tried out the latest draft version of 3.9.0.
I used the PPA method as identified at the website below:
https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-master

Unfortunately I found that the state of the current development on my system was unusable. Simply adding a Signal Source gave issues additional to the experience with v3.8.2.0. For info, the CMD tab which I was told to expect in v3.8.2. did not show either.

I am now putting this activity aside, but wanted to acknowledge all the efforts to help and support me. Thanks guys!

bobbryce, 2020-11-07:
Post script

The GNU Radio wiki page gives the recommended method for install on Ubuntu-based systems as 'apt install gnuradio', however it seems that the selection of 'software sources' is important. Can anyone recommend which software source needs to be set in order to access the v3.8.2.0 package?

The attached text shows the results produced by selecting two software sources - both produce v3.7.1.
[which_cache.txt](https://github.com/gnuradio/gnuradio/files/5505349/which_cache.txt)

Many thanks

drmpeg, 2020-11-07:
@bobbryce The official Ubuntu repositories are cast in concrete. Whatever release was available from GNU Radio back in April of 2020 (in this case, version 3.8.1.0-rc1) is what will be available for Ubuntu 20.04 forever.

https://packages.ubuntu.com/search?keywords=gnuradio

bobbryce, 2020-11-08:
Appreciate the response. Would you recommend I build from source then?

drmpeg, 2020-11-08:
This is a little contrary to what the official GNU Radio stance is, but yes I do recommend building from source. It's just like learning how to ride a bicycle. Scary when you first start out, but easy in hindsight and you'll never forget how to do it.

However, a source build still requires a clean install. I have a sneaky suspicion you'll still have problems with your current Mint install. There's something bizarrely wrong with it.

Before, I mentioned a clean install of Mint. What I meant was a complete disk wipe by installing Mint from a USB stick (which I assume is how you installed Mint in the first place). I know this is a drastic step and may be quite difficult if you have a lots of existing work to backup. But at this point, it's the best path to success.

I can easily guide you through a source install, but you have to start with a clean slate.

bobbryce, 2020-11-08:
OK drmpg. Turns out this is a historic day anyway, so I'm encouraged to give it a go. 

Yes, I installed from USB stick, so ready to go.  Is there a page of instructions you can guide me to?

drmpeg, 2020-11-08:
We can do it right here. Let me install Linux Mint here, so that I can make sure it goes well.

bobbryce, 2020-11-08:
Thnx drmpg. Done at my end. System rebuilt from scratch using Mint Live USB stick. 
It's 02:00 where I am, so will pick this up again tomorrow.
Thanks again.

drmpeg, 2020-11-08:
Cool. I'll leave the instructions here as I proceed.

drmpeg, 2020-11-08:
Here's the recipe. This installs into a custom prefix instead of /usr/local. This keeps /usr/local nice and tidy and makes deleting and rebuilding GNU Radio very easy.

1) Install dependencies

sudo apt-get install git libboost-all-dev libusb-1.0-0-dev python-mako doxygen python-docutils cmake build-essential libncurses5 libncurses5-dev libudev-dev python-numpy python-setuptools python3-yaml python3-mako python3-requests python3-numpy ccache liborc-0.4-dev liblog4cpp5-dev libfftw3-dev libcppunit-dev libgmp-dev libgsl-dev swig libzmq5 libzmq3-dev libsdl1.2-dev python3-click python3-click-plugins python3-gi-cairo libpangocairo-1.0-0 gir1.2-gtk-3.0 python3-pygccxml qtbase5-dev libqwt-qt5-dev pyqt5-dev-tools python3-matplotlib libcanberra-gtk3-module python3-pytest python3-zmq python3-setuptools libsndfile1-dev graphviz

2) Build UHD. You may not need this, but it doesn't hurt.

cd ~/someworkingdirectory

git clone https://github.com/EttusResearch/uhd.git

git checkout v3.15.0.0 -b tmp

cd host/

mkdir build

cd build/

cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.9.0git ../

make -j$(nproc)

sudo make install

3) Build Volk.

cd ~/someworkingdirectory

git clone --recursive https://github.com/gnuradio/volk.git

cd volk/

mkdir build

cd build/

cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.9.0git ../

make -j$(nproc)

sudo make install

4) Build pybind11.

cd ~/someworkingdirectory

git clone https://github.com/pybind/pybind11.git

cd pybind11/

git checkout v2.5.0 -b tmp

mkdir build

cd build/

cmake -DCMAKE_BUILD_TYPE=Release -DPYBIND11_TEST=OFF ../

make

sudo make install

sudo ldconfig

5) Build GNU Radio 3.9

cd ~/someworkingdirectory

git clone https://github.com/gnuradio/gnuradio.git

cd gnuradio/

mkdir build

cd build/

cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.9.0git ../

make -j$(nproc)

sudo make install

6) At a library configuration file (because we're using a custom install prefix).

cd /etc/ld.so.conf.d

sudo nano gnuradio.conf

**** insert the following line, Ctrl-o to save and Ctrl-x to exit ****

/opt/gnuradio-3.9.0git/lib

sudo ldconfig

7) Add paths to .bashrc.

cd

nano .bashrc

**** insert the following lines, Ctrl-o to save and Ctrl-x to exit ****

export PATH=$PATH:/opt/gnuradio-3.9.0git/bin
export PYTHONPATH=$PYTHONPATH:/opt/gnuradio-3.9.0git/lib/python3/dist-packages

8) Exit that shell and start a new one (so that the .bashrc changes take effect) and try gnuradio-companion.

bobbryce, 2020-11-08:
SUCCESS!

drmpg - can't thank you enough. I was certainly not expecting this level of support, but pleased to feedback that this has worked for me with no issues in running the simple level of flowgraphs that had my system on its knees with the 'alleged' v3.8.1 and v3.8.2 that had been installed earlier.

For info, I have enclosed the following screenshots:-

- Launch window
- Signal Source into Frequency sink - both before and after 'Run'
- As above, plus QT Range
- As above, plus QT Chooser

I also included a complete history of the installation in case it's of any help - there were several 'Notes' and 'Warnings' flew by during the make sections, but they don't seem to have impacted the end result. 
 
A question please - have I now got an 'orphaned' installation - i.e. I presume it is not going to be affected by the normal update process as future enhancements are produced?
I'm guessing that what I probably need is to schedule a repeat of the section of the recipe that pulls and makes the gnuradio.git, however the 'cmake' section explicitly references the version number and I'm not sure how to deal with that going forward.

Just one more time to express my sincere thanks for the effort everyone has put in to assist.
Thanks, Bob

[image]
[image]
[image]
[image]
[image]
[image]
[image]

[Build history.txt](https://github.com/gnuradio/gnuradio/files/5506795/Build.history.txt)

closed 2020-11-08 by bobbryce

drmpeg, 2020-11-08:
Awesome! Thanks for including the build history. It was perfect. The warnings are all expected.

As for your question, you essentially have a developer installation of the latest and greatest GNU Radio. It's up to you to decide when to upgrade as new development happens.

The procedure is:

cd gnuradio

git pull                                        (this fetches any updates)

cd build

make                                         (if there were any CMake changes, cmake will run automatically)

sudo make install

sudo ldconfig

As for the other modules, UHD and pybind11 will never update since we're using a specific tag. Volk may update occasionally, but it's fairly stable. If it does update, same procedure as above in the volk directory.

bobbryce, 2020-11-08:
Epic. Thanks again Ron.
You've put a lot of effort into that script - it might make sense to publish that for the benefit of others?  For sure, I leave that for you to decide.
I'm grateful for all the help. Onwards!