gnuradio/gnuradio #3920: Raise required Thrift version (on master) to 0.13.0

marcusmueller, 2020-11-07:
Problem: ctrlport-thrift doesn't build if we don't use the thrift smart pointer type inside our rpcmanager framework.

On maint-3.8, the solution looks like #2739; code is different for Thrift up to including 0.10 from newer version.

On master, we can avoid all the required version detection by bumping dependency version requirements to 0.13.0, which is available: (2020-11-06)

- fedora 31 (close to EOL): 0.10.0 
- fedora 32: 0.13.0
- Ubuntu 18.04: it's getting colder, someone is writing thrift 0.9.1 on the  walls of their cave
- Ubuntu 20.04: 0.13.0
- Debian testing: 0.13.0
- Debian unstable: 0.13.0
- Opensuse tumbleweed: 0.13.0
- (Gentoo: special)

So, we'd be not supporting stock Thrift on Ubuntu 18.04, which means the ctrlport-thrift component won't work there. I'm OK with that.

marcusmueller, 2020-11-07:
@ryanvolz @ZeroChaos- Will this pose a problem?

ryanvolz, 2020-11-07:
> @ryanvolz @ZeroChaos- Will this pose a problem?

This will be just fine with conda-forge, which has 0.13 and where I haven't compiled with thrift in a long time because of version incompatibilities.

maitbot, 2020-11-21:
See

RPC fixes to allow using Thrift-0.13 #3953 

This adds building with 0.13.0 on master, as well as updates to logging for rpcserver_thrift class.
Hopefully will work with Thrift 0.14.0 as well...

closed 2021-02-18 by mbr0wn

ZeroChaos-, 2021-02-18:
apologies I totally missed this, but yeah it's fine :-)