gnuradio/gnuradio #3940: Fedora 33 build bails due to Boost bind

willcode, 2020-11-13:
A number of blocks use boost::bind. On Fedora 33, this causes
`error: ‘_1’ was not declared in this scope`
where <boost/bind.hpp> is not included.

On the master branch, boost::bind is being replaced, but adding includes seems safer on maint-3.8.

willcode, 2020-11-13:
Adding `<boost/bind.hpp>` to `basic_block.h`  is good enough to get things to build. I started adding the include to individual blocks using `boost::bind()`, but got tired. There are a lot of them.

> diff --git a/gnuradio-runtime/include/gnuradio/basic_block.h b/gnuradio-runtime/include/gnuradio/basic_block.h
> index 1d31e2e9b..6841a4eb3 100644
> --- a/gnuradio-runtime/include/gnuradio/basic_block.h
> +++ b/gnuradio-runtime/include/gnuradio/basic_block.h
> @@ -33,6 +33,7 @@
>  #include <boost/foreach.hpp>
>  #include <boost/function.hpp>
>  #include <boost/thread/condition_variable.hpp>
> +#include <boost/bind.hpp>
>  #include <deque>
>  #include <iostream>
>  #include <map>

willcode, 2020-11-13:
F33 has a patch in their src.rpm that adds `using namespace boost::placeholders;` to each file that uses `boost::bind()`. Should have looked there first.

Why don't we have that fix as a PR somewhere?

This issue is actually the same as https://github.com/gnuradio/gnuradio/issues/3720.

balister, 2020-11-16:
I want to say I have a PR somewhere with these patches ...

https://github.com/balister/meta-sdr/blob/dunfell/recipes-core/gnuradio/gnuradio_git.bb#L225

willcode, 2020-11-16:
Closing, this covered by https://github.com/gnuradio/gnuradio/issues/3720

closed 2020-11-16 by willcode