gnuradio/gnuradio #3950: QT GUI Frequency Sink Traces Map Color Wrong

MountainLogic, 2020-11-17:
Using https://wiki.gnuradio.org/index.php/QT_GUI_Frequency_Sink in 3.8.2.0 (Python 3.7.9) installed from Ports to a Mac running Catalina 10.15.7 (19H2). I am seeing a incorrect color mapping of the scope traces seen in the displayed output vs what is requested in the input with multiple inputs.

STEPS TO REPRODUCE:
[image]
Set Freq Sink to 3 inputs.  
Provide an input to the three input that are different
Set trace 1 to Red
Set trace 2 to Green
Set trace 3 to Blue
Set Legends to On (optional)
Run 

BUG:
[image]
Observe in the legend and on the scope that trace 1 is teal when red was expected
Observe in the legend and on the scope that trace 2 is bright purple when green was expected
Observe in the legend and on the scope that trace 3 is dark purple when blue was expected

MountainLogic, 2020-11-17:
Example flow:
[image]

ChristopheSeg, 2021-01-07:
Could this behaviour be related to using gnuradio themes (may be projector.qss in this case) ?
I don't know how GNURadio handle/modify color in case some theme is used.

marcusmueller, 2024-01-28:
Yep, there's a qss loaded which explicitly overloads these colors

closed 2024-01-28 by marcusmueller