gnuradio/gnuradio #3951: Python3 and 'gr-filter/python/filter/design/filter_design.py'

gvanem, 2020-11-18:
Running this tool shows it's not completely Python3 compatible. When using it, I got an exception:

```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "f:\gv\dx-radio\gnuradio\gv-build\lib\site-packages\gnuradio\filter\filter_design.py", line 777, in design
    self.design_iir()
  File "f:\gv\dx-radio\gnuradio\gv-build\lib\site-packages\gnuradio\filter\filter_design.py", line 905, in design_iir
    except StandardError as e:
NameError: name 'StandardError' is not defined
```

Caused by lines [905](https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/design/filter_design.py#L905) and [917](https://github.com/gnuradio/gnuradio/blob/master/gr-filter/python/filter/design/filter_design.py#L917).

According to [this](https://portingguide.readthedocs.io/en/latest/exceptions.html#the-removed-standarderror), simply replacing `StandardError` with `Exception` should work for Python3.

nickoe, 2021-01-05:
There is indeed some documentation in https://www.python.org/dev/peps/pep-3100/ that this should be replaced as suggested.

marcusmueller, 2024-01-28:
runs now!

closed 2024-01-28 by marcusmueller