gnuradio/gnuradio #3962: AttributeError: module 'pmt' has no attribute 'make_dict'

Rot127, 2020-11-23:
I try to set up a simple graph which ends in a meta file sink.
When I try to run it, it throws this exception.

```
Traceback (most recent call last):
  File "/home/user/SDR/Lesson 2/op.py", line 191, in <module>
    main()
  File "/home/user/SDR/Lesson 2/op.py", line 167, in main
    tb = top_block_cls()
  File "/home/user/SDR/Lesson 2/op.py", line 104, in __init__
    self.blocks_file_meta_sink_0 = blocks.file_meta_sink(gr.sizeof_float*freq_buggets, '', samp_rate, 1, blocks.GR_FILE_FLOAT, False, 1000000, pmt.make_dict(), False)
AttributeError: module 'pmt' has no attribute 'make_dict'

>>> Done (return code 1)
```

I also had to import pmt. Otherwise I got a NameError.

willcode, 2020-11-24:
Could you post the simplest flowgraph that shows this? I just added a meta files sink to a flowgraph under 3.8.2.0, added an "import pmt" block, and it worked.

Rot127, 2020-11-25:
The flow graph looks like this:
[image]
The Python version is 3.8.6

willcode, 2020-11-25:
That works here, adding a file name, reasonable sample rate, etc. My Python version is 3.9.0, but I don't think that should matter here.

Is there any chance this is a PYTHONPATH problem, parts of an old install, etc?

dave-pi, 2020-11-25:
It works for me too. GR 3.8.2.0, Python 3.8.5. If you're unsure about the module you can fire up a python3 interpreter and type:
```
import pmt
pmt.__file__
dir(pmt)
```
which will tell where it is being loaded from and which methods it contains.

Rot127, 2020-12-04:
Sorry for letting you wait. I will try to set up a clean VM in the next days and execute the flow graph again.
Will report back.

Rot127, 2020-12-07:
In the first place the issue was probably the unset `PYTHONPATH` and `LD_LIBRARY_PATH` variable. Because of inexperience with Gnuradio I installed a random `pmt` package with pip and forgot about it. Gnuradio therefore imported this `pmt` package installed by pip.

The solution for this problem is therefore to install Gnuradio and add the environment variables as described here:
https://wiki.gnuradio.org/index.php/ModuleNotFoundError#B._Finding_the_Python_library

Thank you for your help!

closed 2020-12-07 by Rot127