gnuradio/gnuradio #3964: Float to int giving wrong result

varke-cpu, 2020-11-24:
I am using float to int block. For example if float is valued 500.000 , the int ant the output is NOT valued 500, but rather 2e-46 or something (which should not be possible), so float to int is just  saying its doing a conversion, but in reality it is not

dave-pi, 2020-11-24:
I've just tried a very simple flowgraph and seems to work (rounds to nearest integer). Note the Number Sink doesn't work with int inputs due an implementation problem (this is mentioned in the documentation).

[image]

How are you checking the int value?

varke-cpu, 2020-11-25:
Well i am using number sink to see if it works as i expect, but this is because of another problem. I am probing INT signal with probe signal, sampling it with function probe and "level function". Then i am putting this int to delay block. The delay is not working and stays on 0. If i replace that delay with variable with a constant of the save value it works fine, but i need it to work dinamically with int (that is output from my custom python block).

dave-pi, 2020-11-26:
I'm not sure I fully understand your use case. We've established 'Float to Int' seems to be working, which was your initial claim. You are now mentioning that the int is the output of your custom python block and that you are using a probe to sample it. I've tested the Probe Signal with scenario hopefully similar to what you have and it also works OK for me.
In the flowgraph below I'm setting a Constant Source to the value of the QT GUI Entry, and then sampling it with a Probe, and setting the Delay the probe value. It works as expected (when you change the GUI entry value, the delay of the blue signal changes accordingly).

[image]
  
[image]

It might be better if you upload a flowgraph with the specific problem you're having. Also note that the Delay accepts a message to change the delay value, so it might be simpler to use that instead of a probe.

varke-cpu, 2020-11-27:
Well I have found a workaround, but it still does not explain the issue. So i am making a simulation of reflected wave from an airplane. It is rather complex simulator as I am also including environmental reflections as FIR filter, which i have calculated from environmental data using raytracing in other simulator. That is the reason i have to use sampling rate of 220M as the wave is in range of 100Mhz and the environmental reflections computed go as short as 5 samples in this case. In this case reflection from airplane is around 100k samples delayed, which works just fine if i use static delay block, but when replaced with dynamic version it just crashes. Airplane reflection outputs are as follows: gain, phase, delay, doppler shift. If i downsample my signal and use the downsampled signal to 192k everything works fine (obviously delay is then only 20 samples or so, which limits data i get from my simulator, which is not great, as I am using this data later for verification of other stuff). If i divide my dynamic 100k delay by 10 and simply use 10 delay blocks in a chain, with each being only 10k samples it crashes aswell. Although, if I use single dynamic delay that goes up to max 10k, it works fine. I checked the ram and it seems i should have enough. I have 12gb of ram where 130k*64bit should only be 8gb although that shouldnt matter as it easlily works with 130k static delay. Confusing.
[image]

In the picture i am showing workaround that works by doing delay after downsampling, but if the dynamic delay was added in bypassed block it would crash, and with static 130k it works. In final version, gain, phase and doppler are also used on the same signal appropriately, this is setup for debuging the delay.

dave-pi, 2020-11-27:
It might help if you break down the problem into smaller steps. For example have you verified the delay_plane Function Probe is working correctly? You can use a Constant Source with value=delay_plane and send it to a Time Sink to see if it's updating correctly.
If it is, then you can test the Delay with that Function Probe but using a Signal Source/Time Sink instead of your Virtual Source/Virtual Sink, to try and isolate the problem.
Function Probe uses a Python Thread that just calls a function at the Poll Rate specified. You can try and put some debug in there or write something to a log file, to make sure it's working as expected (just have a look into the generated .py and you can easily spot the Threading code).
About the msg interface I suggested it's only been added in GR 3.9 so might not be an option.

ewxl, 2020-11-27:
Can you send the flowgraph where it does not work? In the one you sent above there are no float to int, only int to float.

I don't know what is going on in the Plane Reflection block but I will say that the "color" of the port of your block tells you only how many bytes per item it outputs/inputs. An int32 is aswell as float32 four bytes, so these ports can be connected despite having not matching colors. What it comes down to is what type of 4 byte items is your Plane reflection block really producing? Perhaps double check that the source code really produces int's on those two ports and not anything else?

This example shows what can happen, and your example of getting 2e-46 was the clue:

import numpy as np
a = np.array(500,dtype=np.int32)
a.dtype= np.float32
print(a)
array(7.e-43, dtype=float32)

varke-cpu, 2020-11-30:
Sorry I cannot provide the full flowgraph, because it is for work, but i did test the outputs of delay_plane. It is deffinitely an int32 and also of correct value. I figured out the delay block works, but only if delay_plane is lower than 30k or so. I have checked this signal with xcorr after the shift to confirm it does indeed work. If delay_plane is >30k then the program just crashes. But it wouldnt crash if i used static 130k delay.

ewxl, 2020-11-30:
Ok, then maybe your problem is with the delay-block and not with float to int?

varke-cpu, 2020-12-03:
Well I am suspecting that to be the issue. I am not sure hot it is possible to solve this. I suspected int conversion to be the issue here becuase when i checked it with number sink it gave wrong answer, so most obvious explanation was that float to int is wrong, but apparently its delay that does not work if using large dynamic integer. Although it works with static just fine.

ewxl, 2020-12-03:
Something else to note is that the divide block will crash your program. I can see that you have a divide-block in your flow, and a single zero in the port 2 will crash your program. This could be something to be on the lookout-for.

Going back to your initial statement: "For example if float is valued 500.000 , the int ant the output is NOT valued 500, but rather 2e-46 or something"

Can you reproduce this in some simple manner? If this do happen, it is as you say, a serious problem, but other can't seem to be able to reproduce it.

varke-cpu, 2020-12-04:
Well there is this problem with number sink representation of ints
[image]
 just like @dave-pi  said. 

And also there is problem with dynamic delays larger than some ammont. If i input dynamically changing int to delay block into the signal chain up that is valued up to 10-20k its ok. But if using static int i can use 130k delay with no problems. If dynamic int goes above ~20k then the whole program just freezes indeffinitely without any explanaiton.

ewxl, 2020-12-04:
Ok, the freeze is probably related to something with the buffers and the streams you have going on. I don't know the limit to what offset you can have, and not knowing the complete graph, I can show you this that will fail for some delay (delay is 2**myinput, and failed for me at about 15-16): 

[image]

That is, two parallell flows must, within some limit, produce the same amount of items, if they differ to much (similar to be out of phase), I think the scheduler freezes because he wants one of the streams to catch up. Might this be the case for your total graph?

The QT GUI Number sink and int, as dave-pi first replied, "I've just tried a very simple flowgraph and seems to work (rounds to nearest integer). Note the Number Sink doesn't work with int inputs due an implementation problem (this is mentioned in the documentation)." Hopefully someone will fix this, but still, already a known issue.

So back to Float to int, it doesn't seem to be the problem here. This issue could be closed IMO.

varke-cpu, 2020-12-04:
Thats great info to know! Now that you explained about the scheduler i think i can try a few workarounds to see if it helps, BUT. Again it doesnt make sense that everything is just fine with static int delay of same size, and crashes if i make it dynamic. 
[image]
If yellow block is static 130k it works (when i enable it ofc). If its dynamic 130k it crashes. If its dynamic 10-20k works. My workaround is to downsample the signal to smaller sampling rate and then shift by a wanted delay, but that sort of makes my simulation less useful for further experimentation, as I am using the simulation to asses what sort of hardware I will need for a project. Assuming my hardware goes to 2Msps i want to see what sort of data i am loosing by using that sampling rate as there are also FIR filters in the simulation that are in range of 1-20 of samples in 220Msps.

My main concern here is that static 130k delay works and dynamic 110k in the same block outright crashes which doesnt make much sense.

ewxl, 2020-12-04:
Someone with more knowledge on the scheduler could maybe shed some light on this. Though that is a new issue, I think you should close this one.

closed 2020-12-04 by varke-cpu