gnuradio/gnuradio #3969: Segmentation fault in pkg-config for gnuradio-qtgui.pc - brew - macos

RobertoD91, 2020-11-28:
Installed gnuradio from brew on macos 11.0.1

This command: `pkg-config --list-all` causes a Segmentation fault error if `/usr/local/lib/pkgconfig/gnuradio-qtgui.pc` exist.

If i try `pkg-config gnuradio-qtgui --cflags` or `pkg-config gnuradio-qtgui --libs` i get this error:
`Variable 'PC_LIBS_STR' not defined in '/usr/local/lib/pkgconfig/gnuradio-qtgui.pc'`

[gnuradio-qtgui.pc.txt](https://github.com/gnuradio/gnuradio/files/5611013/gnuradio-qtgui.pc.txt)

rgov, 2020-12-04:
The file is supposed to be configured by CMake here:

https://github.com/gnuradio/gnuradio/blob/7470a7a3771dd90defb826b464dfe62977cb1eb6/gr-qtgui/CMakeLists.txt#L72-L80

Because it passes `@ONLY` to `configure_file`, CMake is only making substitutions of the form `@VAR@` and not `${VAR}`. But the variables it wants to substitute are in `${VAR}` form:

https://github.com/gnuradio/gnuradio/blob/adaa7a265eef17d3b2f9a991e944fe20677b069b/gr-qtgui/gnuradio-qtgui.pc.in#L10-L11

This commit was the problem: https://github.com/gnuradio/gnuradio/commit/76a8de36e49ee81fc23df3c6fc8f935c4257d909#

But the weird thing to me is that [this patch](https://sources.debian.org/patches/gnuradio/3.8.2.0-9/fix-pkg-config/) exists in the Debian sources but it doesn't look like it applies to this codebase? (And why does Debian have *so many* patches that haven't been upstreamed?)

marcusmueller, 2020-12-17:
I'm astonished `pkg-config` segfaults – that **really** shouldn't be happening, no matter how faulty our files are! Thanks for submitting a fix, I'm trying it as I type this.

PR #3981 (2020-12-17): cmake: Fix variable substitution in pkg-config
Variable interpolation is done with configure_file(... @ONLY), so use @VAR@ and not ${VAR}.

I do not have a great way to test this, but I'm pretty sure the way it is now is wrong.

Fixes #3969.
commit 72b36257 (2020-12-17): cmake: Fix variable substitution in pkg-config

Variable interpolation is done with configure_file(... @ONLY), so use @VAR@ and not ${VAR}.

Fixes #3969.
closed 2020-12-17 by marcusmueller