gnuradio/gnuradio #3970: GRC Failed to initialize GTK which causes system error || Manjaro Linux

hakeem7667, 2020-11-28:
I'm running Manjaro 20.2 Deepin with gnu-radio version 3.8.1.0 and it seems like I've got Gtk version 3.0 and 2.0 in my machine.

> $ gnuradio-config-info -v
3.8.1.0

I get a Python System error when running GR companion:

>$ gnuradio-companion
Traceback (most recent call last):
  File "/usr/bin/gnuradio-companion", line 63, in check_gtk
    from gi.repository import Gtk
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "/usr/lib/python3.8/site-packages/gi/importer.py", line 144, in load_module
    importlib.import_module('gi.repository.' + dep.split("-")[0])
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 618, in _load_backward_compatible
  File "/usr/lib/python3.8/site-packages/gi/importer.py", line 145, in load_module
    dynamic_module = load_overrides(introspection_module)
  File "/usr/lib/python3.8/site-packages/gi/overrides/__init__.py", line 118, in load_overrides
    override_mod = importlib.import_module(override_package_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/lib/python3.8/site-packages/gi/overrides/Gdk.py", line 438, in <module>
    initialized, argv = Gdk.init_check(sys.argv)
SystemError: gi.FunctionInfo(init_check) returned a result with an error 

[image]

How can I fix this issue. Thanks

dave-pi, 2020-11-29:
Hi, I've recently tried Ubuntu DDE 20.10 just for kicks and had the exact same problem. I managed to work around that problem but in a very non-ideal way. Just wrap it in a try/except and in the except clause retry the instruction:
```
try:
   Gtk.init_check()
except Exception as err:
   Gtk.init_check()
```
I found that the 2nd call doesn't fail and GRC proceeds to load. It's something with the Gtk Deepin theme and I did try to debug further for a while, but then I just threw the towel. My guess is that there could be other subtle issues so I switched to Kubuntu 20.04.

7keem, 2020-11-29:
Thank you. It worked. I also think that the ideal Linux platform for GRC is Ubuntu too.

closed 2020-11-29 by 7keem