gnuradio/gnuradio #3971: File Source Under Windows

MountainLogic, 2020-11-30:
Two apparent File Source issue.  The fundamental is a 4.1 GB limit to file source under windows 3.8.1.0 (Python 2.7.10).  I've not seen this limit under MacOS.  Is this a feature or a bug?

I've also seen an odd parsing of under scores in window file names on windows that I'm not sure is real or just an odd MyFlow.py generation artifact.

I'm having trouble opening a file with the "File Source" block I recorded with gqrx that are larger than 4 GB.  I have File Source set to vector=1, offset-0, length=0.  I am able to use this file on my Mac GRC 3.8, but not windows GRC 3.8.  I'm running GRC 3.8.0 on windows from the pre-built bins (also got same results after upgrading to the current 'official' 3.8.1 bins with the needed bat file fix).    I found one apparent bug with underscores, but even with working around this (changing underscores to dashes in file names) I still get a fstat failure.  

I'm getting the following in the console output:

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Generating: 'C:\\Users\\FooBar\\Google Drive\\Weather\\SDR\\GR\\wsr88d_001.py'

Executing: C:\Program Files\GNURadio-3.8\gr-python27\python.exe -u C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py

gr::log :ERROR: file_source0 - C:\Users\FooBar\wip\gqrx_20201116_162754_2743499985_19999999_fc.raw: Unknown error
Traceback (most recent call last):
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 230, in <module>
    main()
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 208, in main
    tb = top_block_cls()
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 129, in __init__
    self.blocks_file_source_0 = blocks.file_source(gr.sizeof_gr_complex*1, "C:\Users\FooBar\wip\gqrx_20201116_162754_2743499985_19999999_fc.raw", True, 0, 0)
  File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\blocks\blocks_swig0.py", line 1100, in make
    return _blocks_swig0.file_source_make(itemsize, filename, repeat, offset, len)
RuntimeError: can't fstat file

>>> Done (return code 1)
 
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

In the wsr88d_001.py line 182 I'm seeing double back slashes and missing underscores

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
self.blocks_file_source_0 = blocks.file_source(gr.sizeof_gr_complex*1, "C:/Users/FooBar/wip/gqrx-20201116-162754-2743499985-19999999-fc.raw", True, 0, 0)
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

When I change the file name to use dashes inplace of underscores line 182 seems to have the spaces correctly filled with dashes:

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
 self.blocks_file_source_0 = blocks.file_source(gr.sizeof_gr_complex*1, "C:/Users/FooBar/wip/gqrx-20201116-162754-2743499985-19999999-fc.raw", True, 0, 0)
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

however I get a different error message from the console:

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Generating: 'C:\\Users\\FooBar\\Google Drive\\Weather\\SDR\\GR\\wsr88d_001.py'

Executing: C:\Program Files\GNURadio-3.8\gr-python27\python.exe -u C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py

gr::log :ERROR: file_source0 - C:/Users/FooBar/wip/gqrx-20201116-162754-2743499985-19999999-fc.raw: Unknown error
Traceback (most recent call last):
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 230, in <module>
    main()
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 208, in main
    tb = top_block_cls()
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 129, in __init__
    self.blocks_file_source_0 = blocks.file_source(gr.sizeof_gr_complex*1, "C:/Users/FooBar/wip/gqrx-20201116-162754-2743499985-19999999-fc.raw", True, 0, 0)
  File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\blocks\blocks_swig0.py", line 1100, in make
    return _blocks_swig0.file_source_make(itemsize, filename, repeat, offset, len)
RuntimeError: can't fstat file

>>> Done (return code 1)
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

With 3.8.1 I get the following with a different file:
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>
Generating: 'C:\\Users\\FooBar\\Google Drive\\Weather\\SDR\\GR\\wsr88d_001.py'

Executing: C:\Program Files\GNURadio-3.8\gr-python27\python.exe -u C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py

gr::log :ERROR: file_source0 - C:\Users\FooBar\wip\gqrx_20201021_002311_2738137998_50000000_fc.raw: Unknown error
Traceback (most recent call last):
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 236, in <module>
    main()
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 212, in main
    tb = top_block_cls()
  File "C:\Users\FooBar\Google Drive\Weather\SDR\GR\wsr88d_001.py", line 130, in __init__
    self.blocks_file_source_0_0 = blocks.file_source(gr.sizeof_gr_complex*1, 'C:\\Users\\FooBar\\wip\\gqrx_20201021_002311_2738137998_50000000_fc.raw', True, 0, 0)
  File "C:\Program Files\GNURadio-3.8\lib\site-packages\gnuradio\blocks\blocks_swig0.py", line 1100, in make
    return _blocks_swig0.file_source_make(itemsize, filename, repeat, offset, len)
RuntimeError: can't fstat file

>>> Done (return code 1)
<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

drmpeg, 2020-12-01:
@ryanvolz Can you test this with your Anaconda builds? I think _fstat() in gr-blocks/lib/file_source_impl.cc may need to be _fstat64().

ryanvolz, 2020-12-01:
> @ryanvolz Can you test this with your Anaconda builds? I think _fstat() in gr-blocks/lib/file_source_impl.cc may need to be _fstat64().

I kicked off a conda-forge build with a patch, and we'll see what happens.

ryanvolz, 2020-12-01:
It has gotten past building file_source_impl.cc with this patch:
```
diff --git a/gr-blocks/lib/file_source_impl.cc b/gr-blocks/lib/file_source_impl.cc
index ee8f08d98..73bbda975 100644
--- a/gr-blocks/lib/file_source_impl.cc
+++ b/gr-blocks/lib/file_source_impl.cc
@@ -37,9 +37,9 @@
 #ifdef _MSC_VER
 #define GR_FSEEK _fseeki64
 #define GR_FTELL _ftelli64
-#define GR_FSTAT _fstat
+#define GR_FSTAT _fstati64
 #define GR_FILENO _fileno
-#define GR_STAT _stat
+#define GR_STAT _stati64
 #define S_ISREG(m) (((m)&S_IFMT) == S_IFREG)
 #else
 #define GR_FSEEK fseeko
```

@MountainLogic will you be able to test from a conda package when the build is done, or will I have to go and create a file that's more than 4 GB and test it locally?

MountainLogic, 2020-12-01:
@ryanvolz Happy to test.  Are their directions somewhere for using an anaconda package?

ryanvolz, 2020-12-01:
I have to get around to documenting this on the Wiki, but here's what you'll need in this case given that you already have an Anaconda installation. Open an Anaconda shell (you should have a launcher for that from your Anaconda3 installation, although the key part is that you're in a shell where the `conda` command is recognized). Then:
```
conda create -n grtest
conda activate grtest
conda config --env --add channels conda-forge
conda config --env --add channels ryanvolz/label/bigfile
conda config --env --set channel_priority strict
conda install gnuradio python=3.8
```

That will give you a 3.8.2 installation with the above patch using Python 3.8 in the 'grtest' environment. You should be able to launch `gnuradio-companion` from within that shell or just run an existing GR script with `python`. Let me know if that doesn't work out.

MountainLogic, 2020-12-01:
Thanks.  My Anaconda had become dusty and crufty.  Doing a clean install.  Will then test.

ryanvolz, 2020-12-01:
> Thanks. My Anaconda had become dusty and crufty. Doing a clean install. Will then test.

Then in case it's not too late and this is of interest to you, you can instead install [Miniconda](https://docs.conda.io/en/latest/miniconda.html) just to get the bare bones base environment going if you don't need the whole Anaconda distribution.

MountainLogic, 2020-12-01:
Got almost to the end installing the test package and then got this:

```
python-3.8.6         | 19.2 MB   | ########################################################################################################################################### | 100%
click-plugins-1.1.1  | 9 KB      | ########################################################################################################################################### | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: -
/- menuinst Exception
Traceback (most recent call last):
  File "C:\Users\FooBar\Anaconda3\lib\site-packages\conda\gateways\disk\create.py", line 245, in make_menu
    import menuinst
  File "C:\Users\FooBar\Anaconda3\lib\site-packages\menuinst\__init__.py", line 23, in <module>
    from .win32 import Menu, ShortCut
  File "C:\Users\FooBar\Anaconda3\lib\site-packages\menuinst\win32.py", line 11, in <module>
    import pywintypes
  File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\win32\lib\pywintypes.py", line 136, in <module>
    __import_pywin32_system_module__("pywintypes", globals())
  File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\win32\lib\pywintypes.py", line 64, in __import_pywin32_system_module__
    import _win32sysloader
ImportError: DLL load failed while importing _win32sysloader: %1 is not a valid Win32 application.menuinst Exception
Traceback (most recent call last):
  File "C:\Users\FooBar\Anaconda3\lib\site-packages\conda\gateways\disk\create.py", line 245, in make_menu
    import menuinst
  File "C:\Users\FooBar\Anaconda3\lib\site-packages\menuinst\__init__.py", line 23, in <module>
    from .win32 import Menu, ShortCut
  File "C:\Users\FooBar\Anaconda3\lib\site-packages\menuinst\win32.py", line 11, in <module>
    import pywintypes
  File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\win32\lib\pywintypes.py", line 136, in <module>
    __import_pywin32_system_module__("pywintypes", globals())
  File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\win32\lib\pywintypes.py", line 64, in __import_pywin32_system_module__
    import _win32sysloader
ImportError: DLL load failed while importing _win32sysloader: %1 is not a valid Win32 applicatiodone
```

MountainLogic, 2020-12-01:
Probibly me having littl eexperiance under anaconda.  Was able to launch Navigator from inside the anaconda shell.  I selected the grtest environment and then got this message:

```
Traceback (most recent call last):
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\runpy.py", line 194, in _run_module_as_main
return _run_code(code, main_globals, None,
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\runpy.py", line 87, in _run_code
exec(code, run_globals)
File "C:\Users\FooBar\Anaconda3\envs\grtest\Library\bin\gnuradio-companion.exe\__main__.py", line 99, in 
File "C:\Users\FooBar\Anaconda3\envs\grtest\Library\bin\gnuradio-companion.exe\__main__.py", line 73, in check_gnuradio_import
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\gr\__init__.py", line 46, in 
from .top_block import *
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\gr\top_block.py", line 32, in 
from .hier_block2 import hier_block2
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\gr\hier_block2.py", line 26, in 
import pmt
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\pmt\__init__.py", line 61, in 
from .pmt_to_python import pmt_to_python as to_python
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\pmt\pmt_to_python.py", line 23, in 
import numpy
File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\numpy\__init__.py", line 138, in 
from . import _distributor_init
File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\numpy\_distributor_init.py", line 26, in 
WinDLL(os.path.abspath(filename))
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\ctypes\__init__.py", line 373, in __init__
self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application
```

ryanvolz, 2020-12-01:
Grrr... that's a stumper. What's the output of `conda list -n grtest`?

It looks like it might be related to creating Start menu entries (menuinst)? If that's the case, you might be able to avoid this by installing `gnuradio-core` instead of `gnuradio` and foregoing use of GRC. (`gnuradio` is a meta-package that installs `gnuradio-core` along with `gnuradio-grc` and other components that you might not need for this.)

The only other thing that I can think to try is using a different base conda environment like Miniconda or [Miniforge](https://github.com/conda-forge/miniforge).

ryanvolz, 2020-12-01:
Now I see from your last traceback that it seems to be loading numpy from outside of the conda environment. Perhaps that didn't get installed correctly in grtest, or the environment is not activated correctly. `conda list -n grtest` will tell us more.

MountainLogic, 2020-12-01:
```
conda list -n grtest
# packages in environment at C:\Users\FooBar\Anaconda3\envs\grtest:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        3.38.0               h5728ca3_1    conda-forge
atk-1.0                   2.36.0               h07fcd51_4    conda-forge
boost-cpp                 1.72.0               h54f0996_4    conda-forge
brotlipy                  0.7.0           py38hab1e662_1001    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2020.11.8            h5b45459_0    conda-forge
cairo                     1.16.0            hd28d34b_1006    conda-forge
certifi                   2020.11.8        py38haa244fe_0    conda-forge
cffi                      1.14.4           py38hd8c33c5_1    conda-forge
chardet                   3.0.4           py38h9bdc248_1008    conda-forge
click                     7.1.2              pyh9f0ad1d_0    conda-forge
click-plugins             1.1.1                      py_0    conda-forge
codec2                    0.9.2                h301d43c_1    conda-forge
cryptography              3.2.1            py38hd8c33c5_0    conda-forge
epoxy                     1.5.4                h8d14728_4    conda-forge
fftw                      3.3.8           nompi_hd3ad3c4_1114    conda-forge
freetype                  2.10.4               h546665d_0    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gdk-pixbuf                2.42.0               hfe18499_1    conda-forge
gettext                   0.19.8.1          hfbb10ce_1004    conda-forge
glib                      2.66.3               h0e60522_0    conda-forge
gnuradio                  3.8.2.0          py38he784ffc_4    ryanvolz/label/bigfile
gnuradio-core             3.8.2.0          py38h206d40c_4    ryanvolz/label/bigfile
gnuradio-grc              3.8.2.0          py38h7f29ce3_4    ryanvolz/label/bigfile
gnuradio-qtgui            3.8.2.0          py38h146eef8_4    ryanvolz/label/bigfile
gnuradio-uhd              3.8.2.0          py38hfdd186c_4    ryanvolz/label/bigfile
gnuradio-video-sdl        3.8.2.0          py38h7f29ce3_4    ryanvolz/label/bigfile
gnuradio-zeromq           3.8.2.0          py38h2bc9259_4    ryanvolz/label/bigfile
gobject-introspection     1.66.1           py38hbffe425_3    conda-forge
gsl                       2.6                  hdfb1a43_0    conda-forge
gtk3                      3.24.23              he3dc4e2_2    conda-forge
hicolor-icon-theme        0.17                 h57928b3_2    conda-forge
icu                       67.1                 h33f27b4_0    conda-forge
idna                      2.10               pyh9f0ad1d_0    conda-forge
intel-openmp              2020.3             h57928b3_311    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
libblas                   3.8.0                    21_mkl    conda-forge
libcblas                  3.8.0                    21_mkl    conda-forge
libclang                  10.0.1          default_hf44288c_1    conda-forge
libffi                    3.2.1             ha925a31_1007    conda-forge
libglib                   2.66.3               h35efcdc_0    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
liblapack                 3.8.0                    21_mkl    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
librsvg                   2.50.2               h1f06ebe_1    conda-forge
libsodium                 1.0.18               h8d14728_1    conda-forge
libtiff                   4.1.0                hc10be44_6    conda-forge
libusb                    1.0.23               ha925a31_3    conda-forge
libxml2                   2.9.10               h1006b36_2    conda-forge
libxslt                   1.1.33               h579f668_1    conda-forge
log4cpp                   1.1.3             ha925a31_1002    conda-forge
lxml                      4.6.2            py38h292cb97_0    conda-forge
lz4-c                     1.9.2                h62dcd97_2    conda-forge
m2w64-gcc-libgfortran     5.3.0                         6
m2w64-gcc-libs            5.3.0                         7
m2w64-gcc-libs-core       5.3.0                         7
m2w64-gmp                 6.1.0                         2
m2w64-libwinpthread-git   5.0.0.4634.697f757               2
mako                      1.1.3              pyh9f0ad1d_0    conda-forge
markupsafe                1.1.1            py38hab1e662_2    conda-forge
menuinst                  1.4.16           py38h32f6830_1    conda-forge
mkl                       2020.4             hb70f87d_311    conda-forge
mpir                      3.0.0             he025d50_1002    conda-forge
msys2-conda-epoch         20160418                      1
numpy                     1.19.4           py38h0cc643e_1    conda-forge
[... 15 lines cut ...]
python_abi                3.8                      1_cp38    conda-forge
pywin32                   228              py38h1e8a9f7_0    conda-forge
pyyaml                    5.3.1            py38hab1e662_1    conda-forge
pyzmq                     20.0.0           py38h758134e_0    conda-forge
qt                        5.12.9               hb2cf2c5_0    conda-forge
qwt                       6.1.5                h265ddaf_0    conda-forge
requests                  2.25.0             pyhd3deb0d_0    conda-forge
[... cut ...]
win_inet_pton             1.1.0            py38h32f6830_1    conda-forge
wincertstore              0.2             py38h32f6830_1005    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                he774522_0    conda-forge
zeromq                    4.3.2                ha925a31_4    conda-forge
zlib                      1.2.11            h62dcd97_1010    conda-forge
zstd                      1.4.5                h1f3a1b7_2    conda-forge
```

ryanvolz, 2020-12-02:
That environment looks fine, but I do wonder if it failed some final installation step and is subtly broken. Does anything besides GRC work? My hope is that you could at least run a gnuradio Python script with the current state. You could use GRC from outside the conda environment to write the Python script and then just run that inside the environment.

MountainLogic, 2020-12-02:
WIll try.

Also in Navigator I went to /environments tab/grtest and clicked on the green arrow to open a  Termial.  Then cd to env\grtest and ran gnuradio-companion.exe and go the following:

```
(grtest) C:\Users\FooBar\Anaconda3\envs\grtest\Scripts>gnuradio-companion.exe
Traceback (most recent call last):
  File "C:\Users\FooBar\Anaconda3\envs\grtest\Scripts\gnuradio-companion-script.py", line 10, in <module>
    sys.exit(main())
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\grc\main.py", line 44, in main
    from gnuradio import gr
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\gr\__init__.py", line 46, in <module>
    from .top_block import *
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\gr\top_block.py", line 32, in <module>
    from .hier_block2 import hier_block2
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\gr\hier_block2.py", line 26, in <module>
    import pmt
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\pmt\__init__.py", line 61, in <module>
    from .pmt_to_python import pmt_to_python as to_python
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\pmt\pmt_to_python.py", line 23, in <module>
    import numpy
  File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\numpy\__init__.py", line 138, in <module>
    from . import _distributor_init
  File "C:\Users\FooBar\AppData\Roaming\Python\Python38\site-packages\numpy\_distributor_init.py", line 26, in <module>
    WinDLL(os.path.abspath(filename))
  File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\ctypes\__init__.py", line 373, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 193] %1 is not a valid Win32 application

(grtest) C:\Users\FooBar\Anaconda3\envs\grtest\Scripts> 
```

MountainLogic, 2020-12-02:
What I'm guessing as pure exe/pyton apps 2to3 pip all seem to run and complain that I'm lacking command line argument, fine.  Anything that seems gr based fails:

```
C:\Users\FooBar\Anaconda3\envs\grtest\Scripts>gr_filter_design.exe                                                    
Traceback (most recent call last):                                                                                        
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\filter\__init__.py", line 31, in <module>    
    from .filter_swig import *                                                                                            
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\filter\filter_swig.py", line 13, in <module> 
    from . import _filter_swig                                                                                          
ImportError: DLL load failed while importing _filter_swig: The specified module could not be found.                                                                                                   
During handling of the above exception, another exception occurred:                                                                                                                                                                             

Traceback (most recent call last):                                                                                        
File "C:\Users\FooBar\Anaconda3\envs\grtest\Scripts\gr_filter_design-script.py", line 6, in <module>                
    from gnuradio.filter.filter_design import main                                                                        
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\filter\__init__.py", line 35, in <module>    
    from .filter_swig import *                                                                                            
File "C:\Users\FooBar\Anaconda3\envs\grtest\lib\site-packages\gnuradio\filter\filter_swig.py", line 13, in <module> 
    from . import _filter_swig                                                                                      
    ImportError: DLL load failed while importing _filter_swig: The specified module could not be found.  
```

I want to be helpful with testing, but if my environment is so fubar'ed, I don't want to waste your time fixing my bad install.  However, I'm still game to try.

ryanvolz, 2020-12-02:
Those are a different error at least, one that points to the environment not being activated/installed correctly since those imports all worked in testing the build. I don't want to waste your time either; feel free to keep trying things, but I'll probably try to give this a spin tomorrow to see if I can read the large files and progress with the actual bug.

MountainLogic, 2020-12-02:
Magic just happened.  Now able to launch and run gnuradio-companion.  Seems to be able to run and open large file just fine.  Normally I hate magic, but today, I'll take it.  

Thanks so much for your help and the fast bug fix.

closed 2020-12-18 by drmpeg