gnuradio/gnuradio #3973: Coding-modules in gnuradio

ewxl, 2020-12-02:
I'm seeing inconsistencies in gnuradio between modules handling coding. Between gr-fec, gr-trellis and gr-digital actually. These modules use polynomials, usually named mask. These are number-based and the binary positions of 1's tell you what polynomial is used. Between modules, these are reciprocal (reversed). I would like to see a gnuradio where this is done in the same way, and I would also like to see it expressed as the exponents used instead of a mask. Under the hood, of course it makes sense that they are used as integers, but to an end user in the workflow, I think it should be expressed mathematically instead, like this:

Example:
p(x) = x^5 + x^3 + x^0 would be like this:
[5,3,0] and would be converted to 0x29.

Blocks affected: Scrambler, Descrambler, Additive Scrambler, CC Coder Definition, and probably more.

The Viterbi and Viterbi Combo block uses fsm to create it's parameters, and this could be abstracted to be easier to use.

I would be happy to dig into this and fix it after a discussion here.

willcode, 2020-12-02:
As much fun as it is to do reverse hex/bit conversions in your head, something that looks like taps would be great. If GRC input is the main concern, yml files could contain code to do the conversion, allowing either a taps format or hex format. Allowing forward/reverse could be more complicated.

ewxl, 2020-12-02:
> As much fun as it is to do reverse hex/bit conversions in your head, something that looks like taps would be great. If GRC input is the main concern, yml files could contain code to do the conversion, allowing either a taps format or hex format. Allowing forward/reverse could be more complicated.

I think this depends on what GR goal is here, there is a concern when coding in python and C++ aswell, using these blocks that is. One could suggest using a conversion tool like the one I implemented here: https://github.com/gnuradio/gnuradio/blob/e3a8cbff1ac3fd8203bf082ecbe727a6e5302416/gr-digital/python/digital/utils/lfsr.py

ewxl, 2020-12-02:
It is actually not even consistent within gr-digital. Now that I investigate, GLFSR does not use the same semantics as the blocks that use (F)LFSR. 

[1]: Mask has to be entered reciprocal in either for these two to produce the same sequence.
[2]: GLFSR also has the x^0 term implicit, so masks aren't even in the same format.

willcode, 2020-12-02:
@ewxl You're right that the API should be consistent in C++ too.

In your example, could `(0b11001,7,1,0)` be `(0b11001,(7,1,0))` to keep the taps/poly portion as a separate arg?

WRT your LFSR example, `reduce((map(lambda)) ...)` :-) we could splurge on multiple lines of Python in whatever the solution ends up being.

ewxl, 2020-12-02:
> @ewxl You're right that the API should be consistent in C++ too.
>
> In your example, could `(0b11001,7,1,0)` be `(0b11001,(7,1,0))` to keep the taps/poly portion as a separate arg?
>
> WRT your LFSR [...]

Of course, just a bad habit from my side ;D

ewxl, 2020-12-03:
I will begin a PR for this.

ewxl, 2020-12-08:
As Daniel Estevez commented in another thread, there is this problem with conventions. I have thought about this a little bit and this is my proposal:

With focus on inheriting methods used in math and being easy to use GnuRadio, the polynomials and seeds could be expressed like this example:

An LFSR of p(x) = x⁶ + x +1, as [6, 1, 0] and a start value of [0,0,0,0,0,1] should produce the sequence [0,0,0,0,0,1,0,0,0,0,1,1]

I'm also suggesting that seed-values should be written in a form of a list or a string, "000001".

[1]: The polynomial can be tested with `np.convolve(sequence, [1,0,0,0,0,1,1])%2`
where the polynomial is easily translated from [1,0,0,0,0,1,1] to p(x), the taps corresponds to the terms in the same order as your write the polynomial.

[2]: The start value of [0,0,0,0,0,1] should produce a sequence that starts with this sequence.

[TC Synchronization and Channel Coding](https://public.ccsds.org/Pubs/231x0b3.pdf)
CCSDS writes polynomials in this manner. It is not stated in clear how the seed value is stated, but looking at figure 6-1, it is clear that this works the same way as in the ETSI-standard and the example above.

Convention-problems:
[ETSI DVB-S2](https://www.etsi.org/deliver/etsi_en/300400_300499/300421/01.01.02_60/en_300421v010102p.pdf)
ETSI-standard use polynomials written reciprocal compared to above given convention. The do however state the seed in this manner.

There are probably more examples of other conventions, so this is my attempt to try and finalize this discussion about it.

daniestevez, 2020-12-10:
Thanks @ewxl. This looks like a very interesting proposal to simplify the use of these blocks.

However, as I commented in another issue, I'm somewhat concerned about backwards-compatibility. There are many people/flowgraphs out there that know that a G3RUH scrambler is done with a mask of `0x21` and a length of `16` (as a curiosity, this even appears in several research papers). I venture to guess that many people "know" this just because they've copied it from somewhere, rather than really understanding what happens inside the block with the mask and length.

I agree to change the interface to make it much more obvious that the polynomial is `1 + x^12 + x^17`, as we can see in the [James' G3RUH article describing the scrambler](https://www.amsat.org/amsat/articles/g3ruh/109.html). However, I'm also slightly worried that people will need to figure out on their own how to change their flowgraphs to make them work with the new definitions/interface. (Here I'm using the G3RUH scrambler just as an example of a popular scrambler).

I'm wondering if we could provide two separate interfaces: a lower level interface that is just like what we have now, and a higher level interface where we specify the polynomial in a user-friendly way. After all, we will still have the same LFSR implementation, so given the user-friendly polynomial, we still need to compute the mask and register length from that. Maybe it makes sense to allow direct access to those parameters as we currently have.

I think that to make this plan work well, it needs good documentation. As @ewxl already mentioned, there are different conventions for things, so for instance the polynomial for G3RUH that I've written above some people will write "reversed" as `x^17 + x^5 + 1`. I think we need to give some examples and definitions so that it is easier to get some polynomials from a spec and work out what needs to be input into GNU Radio.

Another point, I'm not so sure that I understood @willcode's comment about C++, but I think that the API should be "C++"-friendly. Specifying the polynomial `x^17 + x^5 + 1` as `[17, 5, 0]` works quite well in Python, but maybe it's more C++-friendly to use `1U << 17 | 1U << 5 | 1U << 0`, which is quite similar to what we have now. Doing the conversion in the .yml GRC block would be nice, but I'm not sure how well that would play with the generation of a C++-only flowgraph (the conversion code in the .yml would be Python, I guess).

A final note: the LFRS implementation has an implicit `1` term (or maybe it's `x^d`, where `d` is the degree of the polynomial), so if our representation of polynomials allows the user to enter polynomials without leading coefficient or without independent term, we probably should have some error checking that complains to prevent nonsensical results.

laurentkouassi, 2023-08-25:
Greetings

I'm interested in speaking with you about a business deal I desire to actualize with you. To further explore this possibility, I would like you to contact me via my email address to give you a brief description of my field of work and the project idea. Please contact me using my email address stated below
Regards
Mr. Kouassi Laurent
laurenttkouassi@protonmail.com

marcusmueller, 2024-01-28:
Coming back to assess this issue: Daniel hit the proverbial nail on its head: we can't replace the API with one with a different notation for polynomials :/

So, this gets an exercise in adding new constructors, setters and so on to support a consistent format across a lot of blocks that, in all honesty, have worse problems!

I think what I'm taking from this (and other gr-fec issues) is that to ensure consistency and correctness, as little as feasible code should be copied from GNU Radio 3.x to 4.x, but after sitting down and defining a consistent API, blocks should be written, with correct tests and proper examples. Thanks go out especially to @daniestevez for doing so much for gr-fec lately! 

Anyway, I'm afraid, in the current development status that 3.x-series gr-fec is, there's nothing actionable we can do here :(

closed 2024-01-28 by marcusmueller