gnuradio/gnuradio #3974: ULLVector does not exist

ewxl, 2020-12-02:
https://github.com/gnuradio/gnuradio/blob/a8b25c79083cc9abdd5079f5d6fe83965e8fc03e/gr-fec/python/fec/extended_decoder.py#L110

ULLVector is not defined, so this part of the python-code fails, when supplying an annihilator.

If I try to get passed this, the code for using the annihilation doesn't seem to be working anyway, so the rabbit hole grows just deeper because when reviewing the code in the block conv_bit_corr_bb that is used, i don't know what algorithm was intended here, but I think I know a better solution to this problem that I could try to fix. Thought I might make an issue here first to discuss before doing too much work that might not get pulled anyway. Any thoughts?

willcode, 2020-12-02:
ULLVector does not appear anywhere else in the code (master or maint-3.8), so this looks like dead code.

ewxl, 2020-12-02:
I agree. There is more in gr-fec that fails for me from master, none of the Viterbi Combo-blocks exists anymore, I haven't narrowed down why yet, there is a template for them but when the flowgraph runs it is not able to import them.

marcusmueller, 2024-01-28:
MIght want to remote the `ann` variable and all connected to it altogether.