gnuradio/gnuradio #3982: What CCSDS 7-bit register?

ewxl, 2020-12-04:
https://github.com/gnuradio/gnuradio/blob/a3061b8fe69fb8407d52d7f43cae57796015d185/gr-digital/python/digital/qa_scrambler.py#L41

"CCSDS 7-bit scrambler"

This is referenced and reused in some parts of the gr-digital module, however, what scrambler is it referring to? I've tried googling this but no luck, does anyone know what document at CCSDS describes this?

I would like to verify that this scrambler does what it intends to do, I suspect, humbly if possible, it doesn't. This is why I would like to see the definition of the CCSDS.

dkozel, 2020-12-04:
@daniestevez Any chance you recognize this CCSDS scrambler?

daniestevez, 2020-12-04:
I'm afraid not. This scrambler doesn't ring any bells with me.

As far as I know, there is a single CCSDS scrambler, which is described in the [TM Synchronization and Channel Coding](https://public.ccsds.org/Pubs/131x0b3e1.pdf) blue book, Section 10. This is implemented in gr-satellites with the same "Additive Scrambler" block, but [using different parameters](https://github.com/daniestevez/gr-satellites/blob/59dba2289695b8818f38c3ad8d89473f40d3b80a/python/hier/ccsds_descrambler.py#L35) to those shown above. The output of the scrambler sequence (i.e., the output when the scrambler input is all zeros) starts by `ff 48 0e c0 9a`, as one can see from the blue book.

I ran the scrambler with the values `0x8a` and `0x7f` given above and the output of the scrambling sequence starts by
`fe 08 62 9e 8e 4b 76 6a fc 10 c5 3d 1c 96 ec d5`. I haven't found anything in Google that matches this sequence (sometimes some code that implements the scrambler sequence as a hard-coded array will show up).

I guess that to test the scrambler block these values are as adequate as any other (they are also the default values of the GRC block, by the way). However, the "CCSDS 7-bit scrambler" comment in the code seems misleading to me.

I wonder what the person that originally wrote this had in mind, since I can't map this to a real-world protocol (but of course there are many protocols I don't know well).

willcode, 2020-12-04:
It looks like GR uses this only for BER testing where the exact scrambler doesn't really matter. The comment is probably wrong. I looked around, browsed, the CCSDS docs, etc., and couldn't find anything close either.

ewxl, 2020-12-05:
Thanks for your thoughts on this, what I have been able to confirm:
[1]: 0x8a, length 7 does produce a sequence, but the first bit is flawed, of what seems to be p(x) = x⁶ + x + 1
All these three sets of parameters produce the same sequence and repetition, but only the last one does a good job of telling you what the real polynomial is.
(0x8a,0x6e,7)
(0xc5,0x6e,6)
(0x43,0x2e,5)

[2]: I also wonder what the intention was from the beginning, i believe this to be a mistake.

willcode: I also found it in the packed_header_ofdm files, but I suspect the randomizer to be slightly wrong, possibly only the first bit, but still. 

It is also used for all of the examples of scramblers in gr-digital's qa_scrambler.py. My opinion on this is, that the qa doesn't test whether the correct scrambler-sequence is produced, it only checks that it makes same sequence twice, and of course it will. This is the reason why qa missed the bug that was in the additive_scrambler that someone fixed a while ago. 

Right now, the case is that the additive_scrambler works just fine when you put in the correct parameters, but the original qa has no idea if they are correct. This will be partly fixed by the qa in my PR on the scrambler improvement, however I consider the old qa to be flawed. It should be considered to update or remove it. This will be the case for my soon to come PR that updates consistency in gr-digital.

daniestevez, 2020-12-05:
> but the first bit is flawed

I wouldn't say the first bit is flawed. We're loading up `0x7e` into the LFSR, so the first output byte of the scrambling sequence is going to be `0xfe`, just replicating the LFSR contents in reverse order (the LFRS has 8 bits, even though the "length" parameter is 7 because, well, conventions).

> All these three sets of parameters produce the same sequence and repetition

I didn't check this, but not sure I follow. In my mind a scrambler with "length" `l` will produce a sequence of length `2**(l+1)-1` bits, provided that the polynomial is primitive. So maybe the thing here is that the polynomial corresponding to `0x8a` is not primitive (again, there is an arguably weird convention in place for how to specify the polynomials here).

> that the qa doesn't test whether the correct scrambler-sequence is produced

I think that a good approach here would be to instantiate the scrambler with the parameters corresponding to a well-known scrambling sequence, such as the CCSDS scrambler, and then check that its output matches an array that is hardcoded with that sequence.

ewxl, 2020-12-05:
Yes, you are right, the first bit is not flawed, i just wasn't able at first to produce the sequence with the other parameters, but that did work out finally.

I agree the problem here is perhaps conventions. Is there an obvious reason why you supply both mask and length of the register, since length can be deduced from the mask, if the mask represents the real polynomial?

If you try those the sets you'll see that the periodicity is 2**6-1, they all produce this sequence over and over:
011101100110101011111100000100001100010100111101000111001001011
Which is from [6,1,0], indeed a 7-bit scrambler, and primitive aswell.

Here is some code you can try it with:

%pylab notebook
import numpy as np
from gnuradio import digital
db = lambda x: "".join(map(str,x))

for mask,seed,rlen in [(0x8a,0x6e,7),(0xc5,0x6e,6),(0x43,0x2e,5)]:
    s = digital.lfsr(mask, seed, rlen)
    seq = [s.next_bit() for _ in range(2**6-1)]
    print(db(seq))
    seq = [s.next_bit() for _ in range(2**6-1)]
    print(db(seq))
    seq = np.array(seq,dtype=np.float32)*2-1
    res = np.correlate(seq,seq,mode="full")
    figure()
    plot(res)
    show()

daniestevez, 2020-12-05:
> I agree the problem here is perhaps conventions. Is there an obvious reason why you supply both mask and length of the register, since length can be deduced from the mask, if the mask represents the real polynomial?

The mask contains bits `a1 a2 ... aL` from the polynomial `p(z) = 1 + a1*z + a2*z^2 + ... aL*z^L`, so there is no way to deduce the length from the value of the mask. It may well happen that `a1 = 0`.

As I said, the conventions are quite tricky. I have [this old post](https://destevez.net/2016/05/scramblers-and-their-implementation-in-gnuradio/) that I mainly wrote to explain to myself what are the conventions. I need to come back to it any time some question about the scramblers pops up.

daniestevez, 2020-12-05:
Regarding your three examples that give the same output, this is what happens:

* `(0x8a,0x6e,7)`. This is basically the polynomial `x^7+x^5+x+1`. This is a bit weird since you are using a length of 7 for a polynomial of degree 7. It would be enough to use a length of 6.
* `(0xc5,0x6e,6)`. Here I'm guessing but have not checked that a mask of `0x45` should give the same output, because with `0xc5` you are putting one bit of the mask outside the shift register width. The polynomial is again `x^7+x^5+x+1`, if we ignore this extra bit in `0xc5`.
* `(0x43,0x2e,5)`. Here we have the polynomial `x^6+x^5+1`.

Now, it happens that `x^7+x^5+x+1 = (x+1)*(x^6+x^5+1)`. My knowledge of the theory of LFSRs is a bit rusty, but I'm guessing that this means that for some of the initial register values the output with the polynomial  `x^7+x^5+x+1` will match the output with the polynomial `(x^6+x^5+1)`.

daniestevez, 2020-12-05:
By the way, more care with conventions. There are people that define the polynomial like me and there are people that reverse it, by doing `x^deg(p) * p(1/x)`.

ewxl, 2020-12-06:
I know the conventions are divided, that is really not my concern in this issue, however I do think the project should choose one and have it be consistent in the project. (the glfsr and lfsr in gr-digital are not the same)

I read your post, that was really good, I can tell you made the same trip I did in those blocks to determine what they do.

> The mask contains bits `a1 a2 ... aL` from the polynomial `p(z) = 1 + a1*z + a2*z^2 + ... aL*z^L`, so there is no way to deduce the length from the value of the mask. It may well happen that `a1 = 0`.

I would say the contrary, if a1 = 0, the degree of the polynomial decreases? The terms leading or trailing that are zero, will do nothing with the output-bitstream, as in the 0x8a case, and can therefore be "trimmed". 

In that post you do make exactly my point actually, that length is deg(p)-1.

I think I have got all the answers I need, I will post a PR that you hopefully will like, that makes it alot easier to use these blocks.

daniestevez, 2020-12-06:
> I would say the contrary, if a1 = 0, you still have the constant term. If aL = 0, the degree of the polynomial just decreases? The terms leading or trailing that are zero, will do nothing with the output-bitstream, as in the 0x8a case, and can therefore be "trimmed".

I don't agree with this. For a real-world example, the mask for the G3RUH asynchronous scrambler is `0x21`. Can you implement this without knowledge of the polynomial degree or shift register length? (i.e., can you guess the shift register length just from knowing the mask?).

ewxl, 2020-12-06:
> I don't agree with this. For a real-world example, the mask for the G3RUH asynchronous scrambler is `0x21`. Can you implement this without knowledge of the polynomial degree or shift register [...]

Not if the mask is incomplete, but what my PR is going to suggest is that the polynomial is expressed, and not a combination of mask and length.

daniestevez, 2020-12-06:
The problem with changing the interface is that it would break backwards-compatibility. Maybe not a problem, since GNU Radio 3.9 will already be breaking in several aspects. But if we're going to change this, we should probably give it a good thought on what to change it to, given the different conventions out there (for instance the CCSDS blue book uses the reverse polynomial with respect to what I usually write).

ewxl, 2020-12-06:
I agree, that will probably be the biggest challenge, and I don't have a really good suggestion there yet. Usually, math isn't possible to be interpreted differently, still in this case it seems that's exactly what happens.

ValZapod, 2021-01-08:
> I've tried googling this but no luck

Second page. https://lists.gnu.org/archive/html/discuss-gnuradio/2010-01/msg00547.html (and replies), also here the implemenation in C++ that did not get accepted, apparently, as I can see only in python. https://www.mail-archive.com/patch-gnuradio@gnu.org/msg00180.html

This is some hacky for internal use only scrambler? Dunno.

ValZapod, 2021-01-08:
> for instance the CCSDS blue book uses the reverse polynomial with respect to what I usually write

Mmm... from abobe C++ implementation above:

```
some caveats:
- Polynom specification for both types are different.
For Gal', one only has to use the binary representation (e.g. x^4+x^3+1 => 
2^4+2^3+1 = 0x19) and shift it one bit left ( 0x13>>1 = 0x9).
For Fib, the following has to be done: Reverse the binary representation (0x19 
= 11001 => 10011), strip the highest bit. The resulting parameters for the 
lfsr are (mask=0x3, len=(degree-1)=3). (Yes, this is somewhat contrary to the 
comments in gri_lfsr.h. IMHO, the documentation is wrong/misleading. Most 
noteworthy, the len paramater is not the register len, but the bitshift.) 
```
etc.

ewxl, 2021-01-09:
Interesting history, thanks! Let's see if we can get it better this time.

closed 2021-01-09 by ewxl