gnuradio/gnuradio #3983: GNURadio make from source failed

ShaoPaoLao, 2020-12-05:
Hi,
I'm trying to install GNURadio 3.8 after successful installation of UHD 4.0
but I always got the problem like below:

> _/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_i2c_iface_eeprom16(PyObject*, PyObject*)’:
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:30566:33: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<boost::shared_ptr<uhd::i2c_iface> >’ and ‘uhd::i2c_iface::sptr {aka std::shared_ptr<uhd::i2c_iface>}’)
       result = (arg1)->eeprom16();
                                 ^
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:32:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = boost::shared_ptr<uhd::i2c_iface>]
   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
                     ^~~~~~~~
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:32:21: note:   no known conversion for argument 1 from ‘uhd::i2c_iface::sptr {aka std::shared_ptr<uhd::i2c_iface>}’ to ‘const SwigValueWrapper<boost::shared_ptr<uhd::i2c_iface> >&’
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:36:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = boost::shared_ptr<uhd::i2c_iface>]
   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
                     ^~~~~~~~
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:36:21: note:   no known conversion for argument 1 from ‘uhd::i2c_iface::sptr {aka std::shared_ptr<uhd::i2c_iface>}’ to ‘const boost::shared_ptr<uhd::i2c_iface>&’
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:30576:109: error: invalid static_cast from type ‘SwigValueWrapper<boost::shared_ptr<uhd::i2c_iface> >’ to type ‘const sptr& {aka const std::shared_ptr<uhd::i2c_iface>&}’
   resultobj = SWIG_NewPointerObj((new uhd::i2c_iface::sptr(static_cast< const uhd::i2c_iface::sptr& >(result))), SWIGTYPE_p_boost__shared_ptrT_uhd__i2c_iface_t, SWIG_POINTER_OWN |  0 );
                                                                                                             ^
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:1191:89: note: in definition of macro ‘SWIG_NewPointerObj’
 #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
                                                                                         ^~~
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_dboard_iface_sptr_eeprom16(PyObject*, PyObject*)’:
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:36426:34: error: no match for ‘operator=’ (operand types are ‘SwigValueWrapper<boost::shared_ptr<uhd::i2c_iface> >’ and ‘uhd::i2c_iface::sptr {aka std::shared_ptr<uhd::i2c_iface>}’)
       result = (*arg1)->eeprom16();
                                  ^
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:32:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const SwigValueWrapper<T>&) [with T = boost::shared_ptr<uhd::i2c_iface>]
   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
                     ^~~~~~~~
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:32:21: note:   no known conversion for argument 1 from ‘uhd::i2c_iface::sptr {aka std::shared_ptr<uhd::i2c_iface>}’ to ‘const SwigValueWrapper<boost::shared_ptr<uhd::i2c_iface> >&’
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:36:21: note: candidate: SwigValueWrapper<T>& SwigValueWrapper<T>::operator=(const T&) [with T = boost::shared_ptr<uhd::i2c_iface>]
   SwigValueWrapper& operator=(const T& t) { SwigMovePointer tmp(new T(t)); pointer = tmp; return *this; }
                     ^~~~~~~~
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:36:21: note:   no known conversion for argument 1 from ‘uhd::i2c_iface::sptr {aka std::shared_ptr<uhd::i2c_iface>}’ to ‘const boost::shared_ptr<uhd::i2c_iface>&’
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:36436:109: error: invalid static_cast from type ‘SwigValueWrapper<boost::shared_ptr<uhd::i2c_iface> >’ to type ‘const sptr& {aka const std::shared_ptr<uhd::i2c_iface>&}’
   resultobj = SWIG_NewPointerObj((new uhd::i2c_iface::sptr(static_cast< const uhd::i2c_iface::sptr& >(result))), SWIGTYPE_p_boost__shared_ptrT_uhd__i2c_iface_t, SWIG_POINTER_OWN |  0 );
                                                                                                             ^
/home/pc/test/gnuradio/build/gr-uhd/swig/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx:1191:89: note: in definition of macro ‘SWIG_NewPointerObj’
 #define SWIG_NewPointerObj(ptr, type, flags)            SWIG_Python_NewPointerObj(NULL, ptr, type, flags)
gr-uhd/swig/CMakeFiles/uhd_swig.dir/build.make:62: recipe for target 'gr-uhd/swig/CMakeFiles/uhd_swig.dir/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o' failed
make[2]: *** [gr-uhd/swig/CMakeFiles/uhd_swig.dir/CMakeFiles/uhd_swig.dir/uhd_swigPYTHON_wrap.cxx.o] Error 1
CMakeFiles/Makefile2:11163: recipe for target 'gr-uhd/swig/CMakeFiles/uhd_swig.dir/all' failed
make[1]: *** [gr-uhd/swig/CMakeFiles/uhd_swig.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs...._

Could anyone help me with my problem?
My purpose here is using RFNoC with USRP, so I have to install UHD 4.0, GNURadio 3.8 and gr_ettus too.
Thank you for spending time on my topic!

mormj, 2020-12-07:
Hi @ShaoPaoLao - a question like this is better suited for the gnuradio mailing list - or in this case the USRP mailing list.  More people monitor there so you are more likely to get help with your build issue.  Please redirect your question there.  Thanks!

https://wiki.gnuradio.org/index.php/MailingLists

closed 2020-12-07 by mormj

MauriiiM, 2020-12-07:
I'm also running into this exact error. I'm running gnuradio 3.8 and get this same error when I tried updating to UHD-4.0 from UHD-3.15.LTS using `pybombs update`

TSlivede, 2021-07-01:
@mormj Would it be possible to reopen this issue? Even if this is not the best place to ask, it does seem like a possible place to ask, especially as https://github.com/gnuradio/gnuradio/blob/master/README.md mentions this issue tracker as the correct place to ask for bugs. (And a build error after successful configure seems like a bug to me). As a closed issue it looks as if the problem was solved although it wasn't. (I got the same error today.)

@ShaoPaoLao Did you ask on the mailing list? If yes, could you post/link the response here? (Or if you solved the problem yourself, could you mention the solution?)

For reference: I got the error when building gnuradio maint-3.8 from source on ubuntu 18 with UHD 4.0.0.0-154-gb061af4f

ShaoPaoLao, 2021-07-01:
> @mormj Would it be possible to reopen this issue? Even if this is not the best place to ask, it does seem like a possible place to ask, especially as [...]

Hi, it is a long time I haven't seen this problem, I successfully installed and using them now. Maybe this problem comes from: you have tried to install many versions of UHD and GNU Radio on your machine. I don't remember exactly, it seems I followed something likes:
+ Rebuild my ubuntu 18.04 and start installing from a new, clean environment
+ After clone UHD from git, change branch to UHD-4.0, building follow this step: https://files.ettus.com/manual/page_build_guide.html
+ Clone GNU from git and then build following this step: https://wiki.gnuradio.org/index.php/InstallingGR
Don't forget to install all dependencies before building GNU.
Hope you can solve it

TSlivede, 2021-07-02:
Thanks for the fast reply.
I didn't want to spend to much time on this, so I tried building gnuradio maint-3.9, which also didn't work (other error). As it was overdue anyways I upgraded to ubuntu 20 and was then able to build gnuradio maint-3.9 (after rebuilding uhd-4).