gnuradio/gnuradio #3984: Virustotal.com detected 2 new trojans from Win10 GNUradio3.8.2.exe installer.

YalekW5435, 2020-12-06:
Microsoft defender antivirus detected one of them as Ymacco trojan horse, unknown if its a false positive:

[image]

YalekW5435, 2020-12-06:
[image]

The Microsoft defender antivirus detection.

System info can be provided if needed later on.

Tails98, 2020-12-06:
Probably just a false positive!

YalekW5435, 2020-12-06:
wheres the proof at that its a false positive?

noc0lour, 2020-12-06:
This binary is neither distributed from a gnuradio.org domain nor via our releases page. In fact it can stem from various places. Please contact wherever you found this binary if you have doubts about its origin. In general, GNU Radio is FOSS, you are encouraged to request a copy of the source code if you got the binary code from someone else.

closed 2020-12-06 by noc0lour