gnuradio/gnuradio #3986: Input not displayed for vector source when repeat mode is set to ON

V18006, 2020-12-08:
Hello,
I am trying to plot output for vector source of a  complex signal but when I switch my mode to repeat OFF QT GUI is not plotting the output. Is there any mistake in my flow graph?  (when repeat mode is YES ,GUI is plotting the signal)

V18006, 2020-12-08:
file:///home/hp/Pictures/RNom1E.png 
Here is the link to my Flow graph

V18006, 2020-12-08:
/home/hp/Pictures/Link to RNom1E.png

dkozel, 2020-12-08:
Hi @V18006 Neither of those photos posted correctly, they're links to files on your computer.

Your vector is probably getting stuck in a block buffer and not fully processed. When you have repeat turned on the Vector source continues supplying data so the vectors get processed by every block and then consumed by the Vector Sink.

We would like to keep the GitHub issue tracker for tracking feature requests and bug reports. Can you post links to your flowgraphs (hosted on Imgur or similar Image hosting sites) on in the project chat or the email list?
https://wiki.gnuradio.org/index.php/Chat
https://wiki.gnuradio.org/index.php/MailingLists

Hopefully we can help you better there.

closed 2020-12-08 by dkozel