gnuradio/gnuradio #3988: Version 3.8.1.0 no Qt Eye Sink Widget available

Rabobsel, 2020-12-09:
Hi, 
i try to find the Eye Sink Widget in GNU Radio ver 3.8.1.0 but it isn't there.
According to notes in this repository it has been added on February 2020.
Can you explain me how i can add it manually to my installation of GNU Radio under Windows ?
Greetings

duggabe, 2020-12-09:
If you look at https://wiki.gnuradio.org/index.php/QT_GUI_Eye_Sink there is a note that it is added in 3.9 which is the current master branch. Since there are major differences between 3.8 and 3.9 internals, it would not be practical to attempt porting it manually.

Rabobsel, 2020-12-09:
Hi duggabe thx for your answer.
But if i see right the newest Binary version of GNU Radio available for windows is 3.8.1.0.
So this means that version 3.9 can't be installed under Windows yet, or am i getting things wrong ?
Greetings

duggabe, 2020-12-09:
The gnuradio master branch will become version 3.9, but it has not been released yet. Once it is released, a Windows binary version will be created. So, not yet.

closed 2020-12-10 by duggabe