gnuradio/gnuradio #3990: Low performance with UDP source block

daniestevez, 2020-12-10:
Following a discussion with @kng about streaming samples by UDP into gr-satellites, I have discovered that the performance of the UDP source block is quite poor in GNU Radio 3.8.2.0.

To test this, I'm using the following flowgraph that streams IQ `int16_t` samples into itself. At a sample rate of 48ksps it works fine, but at 70ksps I already get several
```
gr::log :WARN: udp_source0 - Too much data; dropping packet.
```

[test_udp.grc.zip](https://github.com/gnuradio/gnuradio/files/5675641/test_udp.grc.zip)

[image]

This happens regardless of whether the UDP source and sink are in the same or different flowgraphs (I'm sharing them in the same flowgraph just for convenience). This happens on an i7-2620M laptop and the CPU is very low, so I don't think it's a problem of CPU usage.

The same simple flowgraph using the UDP source and sink blocks from [gr-grnet](https://github.com/ghostop14/gr-grnet) works fine and is able to stream several Msps without losses (I'm using the built-in 64-bit sequence number headers from gr-grnet to check for losses).

Now, I would swear that the in-tree UDP source was working fine in a previous version of GNU Radio, but I'm not sure if it was on 3.7 or an earlier 3.8. The reason I think this is because as part of my [qo100-groundstation](https://github.com/daniestevez/qo100-groundstation) I have a flowgraph that has a section that gets 600ksps IQ `int16_t` samples by UDP from a LimeSDR running on another machine, and downconverts and filters a 48kHz slice to send it to a snd-aloop device which then I can use from WSJT-X or fldigi. It's been a while since I don't use this, but it was working fine in the past, and now it loses a lot of samples in the UDP source. The problem is that I don't remember if when this worked correctly it was still on 3.7 or if I ever tested this to work fine on some 3.8.

willcode, 2020-12-10:
From `udp_source_impl.cc`:
```
const int udp_source_impl::BUF_SIZE_PAYLOADS =
    gr::prefs::singleton()->get_long("udp_blocks", "buf_size_payloads", 50);
```
See if setting
```
[udp_blocks]
buf_size_payloads = something_higher
```
has any effect.

daniestevez, 2020-12-10:
Fantastic!

I just tried
```
buf_size_payloads = 8192
```
and it improves things to the point that even my 600ksps QO-100 flowgraph runs correctly now. Tomorrow perhaps I'll do more tests to try to measure how fast I can stream and read the code a bit to see if my `8192` makes any sense.

daniestevez, 2020-12-11:
I've just done some more benchmarking by playing with the `buf_size_payloads` value. With `2500` I rarely lose any samples. Even if I disable the throttle block and change the random source by a constant source, the UDP sink will max out a CPU core before the UDP source starts losing.

Using `100` instead of the default `50`, I can already stream at 2Msps with very few loses, whereas with the original `50` I already have loses at 70ksps. With `200`, I can do 5Msps with only occasional loses.

So it seems that increasing the value from the default of `50` gives huge performance gains. Since the memory cost of increasing this value is not too high (1472 bytes multiplied by the value of `buf_size_payloads`), I would suggest increasing the default from 50 to perhaps 200 or at least 100, to avoid people from encountering this problem even at pretty low sample rates.

I've done some research on the git history, and it seems that the value of this variable has been `50` since forever, so perhaps something has changed relatively recently within the scheduler that requires a slightly longer buffer than `50` in order to work properly?

willcode, 2020-12-11:
It would be set low to minimize latency for audio ... no right setting for everyone.

The buffering inside the udp source/sink (maybe other network blocks?) could be optimized a little more. It could use the vmcircbuf internally, rather than implement its own.

daniestevez, 2020-12-11:
Does a lower value improve the latency? From taking a look at the code my understanding was that a lower value only improved the memory footprint, but I might have missed something.

willcode, 2020-12-11:
Right, it's payload_size that affects latency. The scheduling must just not be fast enough to keep up. Don't know what a good default would be.

I'm confused about d_rxbuf being 4*payload_size. What if it's complex? The buffer management could use a redo here to make things simpler and avoid the need for 2 memcpys too.

willcode, 2021-02-27:
@daniestevez Is this one all set?

daniestevez, 2021-02-27:
Yes. This can be closed, so closing...

closed 2021-02-27 by daniestevez

ransaraw98, 2021-11-09:
> Fantastic!
> 
> I just tried
> 
> ```
> buf_size_payloads = 8192
> ```
> 
> and it improves things to the point that even my 600ksps QO-100 flowgraph runs correctly now. Tomorrow perhaps I'll do more tests to try to measure how fast I can stream and read the code a bit to see if my `8192` makes any sense.

Please excuse me I'm a newbee here,
How did you change the variable ? I am using gnuradio companion, where are the source files located? Do I have to build from source with the change included?

Thank you

daniestevez, 2021-11-09:
You should add the following lines to `~/.gnuradio/config.conf`

```
[udp_blocks]
buf_size_payloads = 8192
```

ransaraw98, 2021-11-10:
Thank you for your reply.

I am using gnuradio 3.9.3.0
I can only see the file grc.conf there. (I checked using the terminal also).

I added the lines you mentioned above it did not make any difference for me. Packets kept on dropping even at very low sample rates.

However after reading the UDP source documentation I found out that we have to change `/etc/sysctl.conf` and reboot.
After doing that I could reach 4M sample rate without losing packets.

Thanks.