gnuradio/gnuradio #3997: CI stopped building tests on master

noc0lour, 2020-12-17:
Commit a3061b8fe69fb8407d52d7f43cae57796015d185 broke building tests
Cmake variable `Boost_unit_test_framework_FOUND` is not populated in Boost 1.66.0 on the CI.

Thanks @drmpeg for reporting.

```
-- Boost version: 1.66.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- 
-- Configuring testing-support support...
--   Dependency Boost_unit_test_framework_FOUND = 
--   Disabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
```

willcode, 2021-02-27:
Is this one all set? I only looked at one of the Ubuntu-20.04 CIs.
```
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found version "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework
```

drmpeg, 2021-03-01:
I believe @noc0lour fixed this at some point, but I'm not sure how. The root cause has been exposed in PR https://github.com/gnuradio/gnuradio/pull/4287

Only older versions of CMake are affected.

mormj, 2021-10-12:
Closing this as it no longer seems to be an issue in our CI

closed 2021-10-12 by mormj