gnuradio/gnuradio #4015: Broken PFB Channelizer flowgraph

mormj, 2020-12-21:
[image]

Output from one of the channelizer channels in 3.9
[image]

Output in 3.8 looks fine
[image]

drmpeg, 2020-12-22:
If you use custom taps (low-pass filter), it works okay.

willcode, 2020-12-22:
Replacing the Hierarchical Polyphase Channelizer block with the Polyphase Channelizer block works. Why do we have both of these?

mormj, 2020-12-22:
Not sure the reasoning there.

Taking the lazy approach in the background of trying different commits:
a56c3a7f4908133dc6ceb93ffd3b2f9e1dda8e0e - hosed
1d60888071 — good

mormj, 2020-12-22:
572428babb6db488cc17bfe2ada63ffc23a50d63 appears to be the culprit

willcode, 2020-12-22:
That's what caused the problem addressed in https://github.com/gnuradio/gnuradio/pull/4021 too.

PR #6001 (2022-07-16): filter: pfb channelizer demo fix
# Pull Request Details
Fix from @777arc #5567 
Fixes #4015

## Testing Done

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making this PR](https://github.com/gnuradio/gnuradio/blob/main/CONTRIBUTING.md#dco-signed)
- [x] My code follows the code style of this project. See [GREP1.md](https://github.com/gnuradio/greps/blob/main/grep-0001-coding-guidelines.md).
- [ ] I have [... cut]
closed 2022-07-16 by mormj