gnuradio/gnuradio #4019: Library file versioning

drmpeg, 2020-12-22:
Commit https://github.com/gnuradio/gnuradio/commit/76f05e7585e23541eaab0890b8dc90296903795b has an unintended consequence. The library file names have a random git version. For example, on current master:

-- Building for version: v3.8.0.0-845-g0901fcb5 / 3.10.0git

@marcusmueller @mormj

nickoe, 2020-12-29:
@drmpeg What are the specific implications of this "string" change?

drmpeg, 2020-12-29:
The library files end up with an odd name.

[image]

drmpeg, 2021-01-01:
Also version in gnuradio-config-info and GRC sign-on.

[image]

ryanvolz, 2021-01-08:
I'm guessing that because the version is set by running `git describe` in [GrVersion.cmake](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/GrVersion.cmake), you're getting the old version numbers with a large number of commits because more recent git tags are missing locally on your machine? If the intention is to continue using `git describe` to set a version number per-commit when building from git (which I think makes sense), I don't know a way around this. Perhaps a warning? Somehow forcing the syncing of tags?

The library files ending up with a long name also happen because of `git describe`. [GrMiscUtils.cmake](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/GrMiscUtils.cmake) sets the target `VERSION` to `${VERSION}`, which equals `${GIT_DESCRIBE}`, producing those long names. Perhaps a "--match" argument should be added to the `git describe` call to limit only to "v.*" tags?

drmpeg, 2021-01-08:
Thanks for pointing me in the right direction. It looks like the actual problem is that ```git describe``` is actually telling the truth on the master branch. There just isn't a recent tag on master.

```git describe``` is working correctly on other branches. However, on the maint-3.8 branch, CMakeLists.txt was never updated to change the ```VERSION_PATCH``` to  ```git```, so you don't get version number+commit number from last tag+commit.

So no code needs to be changed. but it may be useful to put some sort of tag on master.

closed 2021-01-09 by drmpeg