gnuradio/gnuradio #4023: Please Help try to install gnuradio 3.7x on Ubuntu 20.04

RaymondZHN, 2020-12-23:
cmake -DCMAKE_BUILD_TYPE=Release -DPYTHON_EXECUTABLE=/usr/bin/python3 ../
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type set to Release.
-- Found Git: /usr/bin/git  
-- Extracting version information from git describe...
CMake Deprecation Warning at CMakeLists.txt:73 (cmake_policy):
  The OLD behavior for policy CMP0026 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:76 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:79 (cmake_policy):
  The OLD behavior for policy CMP0045 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:82 (cmake_policy):
  The OLD behavior for policy CMP0046 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Performing Test HAVE_VISIBILITY_HIDDEN
-- Performing Test HAVE_VISIBILITY_HIDDEN - Success
-- Performing Test HAVE_WARN_SIGN_COMPARE
-- Performing Test HAVE_WARN_SIGN_COMPARE - Success
-- Performing Test HAVE_WARN_ALL
-- Performing Test HAVE_WARN_ALL - Success
-- Performing Test HAVE_WARN_NO_UNINITIALIZED
-- Performing Test HAVE_WARN_NO_UNINITIALIZED - Success
-- Compiler Version: cc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -std=gnu99 -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Building Static Libraries: OFF
[... 798 lines cut ...]
--   * gr-channels
--   * gr-noaa
--   * gr-pager
--   * gr-trellis
--   * gr-video-sdl
--   * gr-vocoder
--   * gr-fcd
--   * gr-wavelet
--   * gr-zeromq
-- 
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * python-support
--   * testing-support
--   * gnuradio-companion
--   * gr-comedi
--   * gr-qtgui
--   * gr-uhd
--   * gr-utils
--   * gr-wxgui
-- 
-- Using install prefix: /usr/local
-- Building for version: 3.7.14.0 / 3.7.14.0
-- Configuring incomplete, errors occurred!

duggabe, 2020-12-23:
Unless you have a specific reason to build from source, you should load the PPA for 3.7
Follow the instructions at https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation

RaymondZHN, 2020-12-23:
> Unless you have a specific reason to build from source, you should load the PPA for 3.7
> Follow the instructions at https://wiki.gnuradio.org/index.php/InstallingGR#Ubuntu_PPA_Installation

I tried common sudo add-apt-repository ppa:gnuradio/gnuradio-releases-3.7 to install 3.7 through PPA but i got 
Latest legacy 3.7 maintenance release of GNU Radio
 More info: https://launchpad.net/~gnuradio/+archive/ubuntu/gnuradio-releases-3.7
Press [ENTER] to continue or Ctrl-c to cancel adding it.

Hit:1 http://us.archive.ubuntu.com/ubuntu focal InRelease
Get:2 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease [114 kB]
Get:3 http://security.ubuntu.com/ubuntu focal-security InRelease [109 kB]      
Ign:4 http://ppa.launchpad.net/gnuradio/gnuradio-releases-3.7/ubuntu focal InRelease
Get:5 http://us.archive.ubuntu.com/ubuntu focal-backports InRelease [101 kB]
Err:6 http://ppa.launchpad.net/gnuradio/gnuradio-releases-3.7/ubuntu focal Release
  404  Not Found [IP: 91.189.95.85 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/gnuradio/gnuradio-releases-3.7/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

closed 2020-12-23 by RaymondZHN