gnuradio/gnuradio #4024: Odd behavior of Max Hold in trace options for QT-GUI Frequency sink

bernienor, 2020-12-23:
I'm using the Max Hold function in trace options of the QT-GUI to test filters. But it sometimes behaves very different from what I expected.

[image]

This only happens when displaying half of the spectrum width. 
[image]

When displaying the full spectrum I get this result, as expected:

[image]

This is the setup I'm using to get these results:
[image]

With the control panel option set to yes in the config tab.
[image]

I think I've located the problem in the code. But as I'm having problems getting the source to compile I thought I'd better document what I've found so far here, in case I fall into another rabbit hole.

I'm looking at the files in the master branch. Last commit is 0901fcb5459c5169017c3073630d7aca158bd626

in file: gnuradio/gr-qtgui/lib/FrequencyDisplayPlot.cc starting from line 379: 
`                for (int64_t point = 0; point < _npoints_in; point++) {
                    if (dataPoints[n][point] < d_min_fft_data[point]) {
                        d_min_fft_data[point] = dataPoints[n][point + _in_index];
                    }
                    if (dataPoints[n][point] > d_max_fft_data[point]) {
                        d_max_fft_data[point] = dataPoints[n][point + _in_index];
                    }

`
I think it should be changed to this:
`                for (int64_t point = 0; point < _npoints_in; point++) {
                    if (dataPoints[n][point + _in_index] < d_min_fft_data[point]) {
                        d_min_fft_data[point] = dataPoints[n][point + _in_index];
                    }
                    if (dataPoints[n][point + _in_index] > d_max_fft_data[point]) {
                        d_max_fft_data[point] = dataPoints[n][point + _in_index];
                    }

`
As the +_in_index will effectively make us look at either the whole array (if looking av the full specter) or look at the last half of it when displaying half the specter. It is defined as shown below.

`    int64_t _in_index = d_half_freq ? _npoints_in : 0;`

I hope this points in the right direction to solve the issue.

drmpeg, 2020-12-23:
Verified here. BTW, you can use a noise source to characterize filters. Like so:

[image]

[image]

drmpeg, 2020-12-23:
Your code fix work good. I'll submit a PR.

bernienor, 2020-12-25:
Thanks!

I finally got the source to compile myself. Not sure if the docs should be updated, since I missed one of the steps described required to compile, the pybind11-dev install. I also missed the libgsm1 install, that one was not listet for Ubuntu 20.04. Not sure if I should just add it to the wiki. Since I'm new to gnuradio I should probably follow the mailinglist for a bit before editing anything on the wiki. :)

Chers!
Bernie

dkozel, 2020-12-26:
Hi Bernie. Thanks for the report! @drmpeg thanks for the fix!

Go ahead and make the wiki change! If you do it as a guest then it gets reviewed. New users and contributors often are the people who notice missing steps like this the best.

mbr0wn, 2021-01-04:
Fixed by https://github.com/gnuradio/gnuradio/pull/4025.

closed 2021-01-04 by mbr0wn