gnuradio/gnuradio #4029: New window API causes backward compatibility issues.

drmpeg, 2020-12-24:
Just documenting an issue discussed on Matrix/IRC. If an old flow graph with any filter block is loaded into GRC built from the latest master or maint-3.9, it will fail with errors due to window parameters defined as ```firdes.WIN_XXX``` (instead of the new ```window.WIN_XXX```).

You can fix this issue by changing the window type in each block and regenerating the flow graph (and then changing the window type back to the original). Or you can edit the .grc file and change the occurrences of ```firdes.WIN_XXX``` to ```window.WIN_XXX```.

A possible fix for this would be to change the dtype for all window parameters to enum instead of raw in the filter block .yml files. However,this would only provide compatibility in one direction (from old to new, but not new to old).

willcode, 2020-12-24:
Slightly better than that ... you can select the same window in GRC and it will be updated. No need to select/save/select.

marcusmueller, 2024-01-28:
I think it was 100% right to document this, but with the practical completion of the 3.8->3.9 transition, it's time to close this. Not something we're likely to ever fix with a new automatism :(

closed 2024-01-28 by marcusmueller