gnuradio/gnuradio #4030: How can I get rid of "Warning: failed to XInitThreads()"?

ShnDanYao, 2020-12-28:
How can I get rid of "Warning: failed to XInitThreads()"?

The generated python file includes those lines:
if __name__ == '__main__':
    import ctypes
    import sys
    if sys.platform.startswith('linux'):
        try:
            x11 = ctypes.cdll.LoadLibrary('libX11.so')
            x11.XInitThreads()
        except:
            print("Warning: failed to XInitThreads()")

Should the "libX11.so"  be "libc.so.6"?
How can I change the template of the python file?

drmpeg, 2020-12-28:
Depends on your system.

If you have no graphics at all, then you should set the ```Generate Options``` to ```No GUI``` in the ```Options``` block of your flow graph. Then XInitThreads() isn't called at all.

If you do have graphics, then install x11.

sudo apt-get install libx11-dev

If you're SSH'ing into a headless system, you may need dbus-x11 in addition to libx11-dev.

sudo apt-get install dbus-x11

ShnDanYao, 2020-12-28:
It worked ! Thanks!

closed 2020-12-28 by ShnDanYao

tuscanred, 2022-03-20:
And it worked for me too; thanks!

blueride80, 2024-03-11:
Worked here as well. Many thanks!