gnuradio/gnuradio #4031: QT GUI Chooser default Labels are blank

duggabe, 2020-12-28:
Leaving the Labels field empty [] yields blank labels in the pull-down
tested with:
Ubuntu 20.04 and 20.10
GRC 3.8.1 and 3.9.0.RC0
The documentation states "Choose the number of options available to your chooser. When the label is left blank, the option will be used as the label. Set the number of options to "list" to enter a single list of options and labels. When the labels is an empty list, the options will be used as the label."
1. import 3.7 flowgraph to 3.8.1 or 3.9.0.RC0 - pull down doesn't work at all
2. using
    Num Options: List
    Options: [0.0, 67.0, 71.9, 74.4, 77.0, 79.7, 82.5, 85.4, 88.5, 91.5, 94.8, 97.4, 100.0, 103.5, 107.2, 110.9, 114.8, 118.8, 123.0, 127.3, 131.8, 136.5, 141.3, 146.2, 151.4, 156.7, 162.2, 167.9, 173.8, 179.9, 186.2, 192.8, 203.5, 210.7, 218.1, 225.7, 233.6, 241.8, 250.3]
    Labels: []
    gives a pull-down list with blank entries.
3. using
    Num Options: List
    Options: [0.0, 67.0, 71.9, 74.4, 77.0, 79.7, 82.5, 85.4, 88.5, 91.5, 94.8, 97.4, 100.0, 103.5, 107.2, 110.9, 114.8, 118.8, 123.0, 127.3, 131.8, 136.5, 141.3, 146.2, 151.4, 156.7, 162.2, 167.9, 173.8, 179.9, 186.2, 192.8, 203.5, 210.7, 218.1, 225.7, 233.6, 241.8, 250.3]
    Labels: ["0.0", "67.0", "71.9", "74.4", "77.0", "79.7", "82.5", "85.4", "88.5", "91.5", "94.8", "97.4", "100.0", "103.5", "107.2", "110.9", "114.8", "118.8", "123.0", "127.3", "131.8", "136.5", "141.3", "146.2", "151.4", "156.7", "162.2", "167.9", "173.8", "179.9", "186.2", "192.8", "203.5", "210.7", "218.1", "225.7", "233.6", "241.8", "250.3"]
    gives the expected results.

nickoe, 2021-01-05:
I can confirm this, tested on maint-3.8.

nickoe, 2021-01-05:
I don't think this has ever worked since the conversion from xml to yaml blocks. It looks like the logic supposed to handle this was removed because it was incomplete and did not parse correctly.

https://github.com/gnuradio/gnuradio/commit/6964b6b6839ef5db63fbfacdd721ae58a75c70a8#diff-a2f4d8d678e7e5589bc17b551998a53e1e00b1d735584aa1c91a70ad977638b3L129

Specifically that line hightlighted with the `if lbl:` does not work as expected. If I revert that patch and disregard the i error it looks like this check does not detect "undefined" entries of all_labels and will there for not replace it with the value from the corresponding entry from all_options.

I am not sure how to do this properly. I am not really familiar with mako templates.

haakov, 2021-01-05:
I haven't had time to test it properly yet and I need to clean it up before submitting a PR, but does this commit fix the problem?
https://github.com/haakov/gnuradio/commit/0f06afb7bde0a0adb798b528878b76ae8957e330

nickoe, 2021-01-06:
@haakov A quicktest shoes it is slightly more broken now. Now no entries are rendered, but I see in the code that the labels are still empty strings.

haakov, 2021-01-06:
Haha alright interesting :) would you mind posting the flowgraph you're using? It seemed to work fine on master but I'll try maint-3.8 and see what happens

nickoe, 2021-01-06:
@haakov I can do it when I get home, but I am just using a new untitled graph with that as the only added block. In the latest test I used string type and filled in a as default a, b and c as options and empty labels.

duggabe, 2021-01-06:
@haakov @nickoe Here is a chooser test you can play with:
https://pastebin.com/Yz1s5vi5

haakov, 2021-01-06:
@duggabe thanks!
@nickoe Right, I only tested with the List option. New suggested fix here: https://github.com/gnuradio/gnuradio/pull/4046

nickoe, 2021-01-06:
@haakov ok, cool. Tested ed8fd82caee6d6f3f0f4da9262547818783ce614. 

It is better and worse at the same time. :)

My test case with the string options asserts unexpectedly with the sting type. I have attached my test grc. 

(I also added the demos from @duggabe in that one.)

On top of that I see that if any entry has a label defined and the others not, then they will render blank. I would expect it to fall back to the value.

haakov, 2021-01-06:
Ah, you're right. I've edited the fix

nickoe, 2021-01-07:
@haakov Cool, even better. No errors at least, and it can "auto" render labels if they are not defined, but _only_ if all of them are not defined. It is a minor issue, but my expectation as a user is that it would sort of use the label args to "override", but in this case all labels need to be defined for them to be used at all.

[image]

haakov, 2021-01-08:
You're not nitpicking at all! :grinning: I've updated the branch again

nickoe, 2021-01-08:
@haakov That looks pretty good to me, but this is getting more and more complex. :D   I noticed that using the "Any" type and just inserting a name, it is interpreted as a variable name, but I guess that is good, but it does not render variable values. I am not really sure what is intended here, but I would probably have expected it to show the value as it does for variable blocks as their content and not the name. But note that I am mixing types pretty badly.

[image]

New example attached.
[qt_chooser_example.grc.txt](https://github.com/gnuradio/gnuradio/files/5785795/qt_chooser_example.grc.txt)

haakov, 2021-01-13:
Right! If you leave the label blank, it will use the (evaluated!) option as the label, so I think that makes sense. 
[image]
Allowing the user to set arbitrary expressions as labels would probably be a matter of changing the `dtype` of the label params to `raw`, which I'm afraid would break some existing flowgraphs, so I'm not sure that's worth it.

nickoe, 2021-01-14:
@haakov I just tried to test on your latest commit, but I can't make it show the value evaluated. I have `samp_rate` in the option and nothing in the label field for that option.

closed 2021-01-20 by mormj