gnuradio/gnuradio #4033: add tcp source

alimamzi, 2020-12-31:
hi dear
there is block tcp source in version 3.7.11 but at the time, there is not that block in version 3.8..i need to have that block in my project
what block could i make block like that or how to add that block in my version
i appreciate in advance

jacobagilbert, 2020-12-31:
What are you trying to use the blocks for? The ZMQ blocks are there to bridge streaming or message blocks across a network.

alimamzi, 2020-12-31:
Thanks for responding....
I need to blocks according to tcp server..as I know I don't run tcp server by zmq block.we could run tcp by zmq but I need to build tcp server in version3.8

jacobagilbert, 2020-12-31:
The TCP source and sink blocks were deprecated in 3.8 due to a wide range of issues, they do not exist for 3.8. The functionality they provided is available in the ZMQ module. 

What functionality are you trying to use these blocks for? Without knowing this it is difficult to help you.

If you can use 3.9 there are some new blocks within the new gr-network module.

alimamzi, 2020-12-31:
I want to have data stream in online mode and send that onto soket for run gps sdr sim which need to have tcp source server
,🙏🙏🙏🙏

jacobagilbert, 2020-12-31:
I'm going to close this because as I mentioned the TCP Source block was intentionally removed in 3.8 so this is not a GR issue.

That said, I am sure me or someone else can help you on either the chat.gnuradio.org matrix server or discuss-gnuradio mailing list. You could also raise this as an issue with whatever GPS simulator you are using.

closed 2020-12-31 by jacobagilbert