gnuradio/gnuradio #4034: qtgui: Number sink average not working?

jaredd, 2020-12-31:
The qtgui number sink has an Average setting that ostensibly sets the alpha for a single-pole iir filter to average inputs before displaying them. Testing this setting against an external single-pole iir average shows a stark difference in behavior. 

Setup: See the flowgraph snapshot below. Gaussian source with amplitude 1 into a square function, feed two number sinks directly, one number sink through a single-pole iir filter block. One directly connected number sink has its Average parameter set to the same value as the alpha parameter in the single-pole iir. All other number sinks have Average parameter set to 0.

[image]

Output: The External average displays the expected value while other number sinks show very different numbers that fluctuate as if no averaging were being applied. That behavior was only expected on the No average sink.

Expected: Both the External average and Average sinks show similar numbers approximately equal to 1 with little fluctuation. The No average sink should show a fluctuating value.

The short video below shows the relative fluctuation of the number sink with internal averaging and no averaging vs external averaging.

https://user-images.githubusercontent.com/3237083/103422990-7a422c00-4b61-11eb-8349-9946b0726b73.mp4

A cursory review of the work function seems like a reasonable implementation so I'm kind of scratching my head.

```bash
$ uname -a
Linux system76-pc 5.8.0-7630-generic #32~1605108806~20.10~7e52b13-Ubuntu SMP Wed Nov 11 19:10:30 UTC  x86_64 GNU/Linux

$ gnuradio-config-info --print-all
/home/innerdev/pybombs
/home/innerdev/pybombs/etc
/home/innerdev/pybombs/etc/gnuradio/conf.d
/home/innerdev/.gnuradio

testing-support;python-support;volk;gnuradio-runtime;gr-ctrlport;gnuradio-companion;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;gr-channels;gr-qtgui;gr-trellis;gr-uhd;gr-utils;gr_modtool;gr-vocoder;gr-wavelet;gr-zeromq
3.8.2.0
clang version 11.0.0-++20200414100631+5c1d1a62e37-1~exp1~20200414201304.3176  
Target: x86_64-pc-linux-gnu 
Thread model: posix 
InstalledDir: /usr/bin
clang version 11.0.0-++20200414100631+5c1d1a62e37-1~exp1~20200414201304.3176  
Target: x86_64-pc-linux-gnu 
Thread model: posix 
InstalledDir: /usr/bin
/usr/lib/ccache/clang-11:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized 
/usr/lib/ccache/clang++-11:::-O2 -g -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
```

willcode, 2020-12-31:
There's a conflict between the GUI's concept of average (settable via middle click menu) and the block's concept of average. The GUI defaults to 0, and the block's value gets overwritten in `number_sink_impl.cc/_gui_update_trigger()`.

The current code reads
```
    float a = d_main_gui->average();
    if (a != d_average) {
        set_average(a);
    }
```
and I think it should be
```
    float a = d_main_gui->average();
    if (a != d_average) {
        d_main_gui->setAverage(a);
    }
```

willcode, 2020-12-31:
Better idea if it works would be to call `d_main_gui->setAverage()` from `number_sink_impl::initialize()`.

jacobagilbert, 2020-12-31:
Your second suggestion works, though any change by calling the blocks `set_average()` function will end up erased when `_gui_update_trigger()` is called.

I think the d_main_gui->setAverage() call needs to go into the blocks `set_average()` function too.

willcode, 2021-01-01:
Variants of this problem could exist in other widgets. I doubt too many people set parameters at runtime via block setters, and also set them via the middle click menu.

jacobagilbert, 2021-01-01:
Yeah, definitely true. It was kinda odd making a check for this.

I think that setters should work though.

closed 2021-01-04 by mbr0wn