gnuradio/gnuradio #4035: Support QApplication command-line args

jaredd, 2021-01-02:
Background:
The current GRC top-level python file uses the parse_args method of argparse.ArgumentParser to check command line args. Any arguments not explicitly defined as top-level parameter in the GRC flowgraph will be flagged as an error. The default QT GUI class [QApplication ](https://doc.qt.io/qt-5/qapplication.html#QApplication)and its parent class [QGuiApplication](https://doc.qt.io/qt-5/qguiapplication.html#QGuiApplication) support a number of command line arguments that can be useful for configuring (or disabling) the Qt application look and feel. 

One specific use case is to run a GUI application on a headless box (e.g. a server) without disabling or removing the GUI elements. By specifying a -platform minimal, or -platform offscreen (-platform being an option to QGuiApplication) the GUI-enabled flowgraph may be executed without requiring a display for those GUI elements, thus reducing the burden of disabling GUI elements in the GRC and, often, maintaining two versions of the flowgraph.

Suggestion:
The relevant line in the current top-level GRC-generated application template reads  https://github.com/gnuradio/gnuradio/blob/d8670ae643f84159367d24b1fb48796336adf457/grc/core/generator/flow_graph.py.mako#L364
This could be changed to 
```python
options, _ = argument_parser().parse_known_args()
```
which will now ignore arguments intended for the QApplication. This change has the disadvantage of removing checks on arguments to the flowgraph that may be mistyped or otherwise extraneous.

nickoe, 2021-01-04:
Sort of related to https://github.com/gnuradio/gnuradio/pull/2650

nickoe, 2021-01-05:
@jaredd Shouldn't one give an explicit list to parse_known_args to pass them on?

jaredd, 2021-01-06:
@nickoe I'm not sure what you're referring to when you say "to pass them on". The parse_known_args method looks for the args that had been configured in the ArgumentParser object prior to its call. You can collect unknown args in the second return value and pass them to the QApplication, that also works. My suggestion was to minimize changed lines. QApplication seems to ignore args it doesn't understand so other than possible mixups if the GR application args somehow match a QApplication arg, the behavior is pretty well defined.

tjeng, 2021-04-30:
I am working on resolving the issue.