gnuradio/gnuradio #4042: pmt: equality/equivalence tests do not work for pair-type objects (including dictionaries)

jacobagilbert, 2021-01-04:
While it is documented correctly, this surprised me today:

```python
>>> import pmt
>>> pmt.eqv(pmt.PMT_EOF, pmt.PMT_EOF)
True                                                                                                      << makes sense
>>> pmt.eqv(pmt.cons(pmt.PMT_NIL, pmt.PMT_NIL), pmt.cons(pmt.PMT_NIL, pmt.PMT_NIL))
False                                                                                                     << ?
>>>
>>> s=pmt.dict_add(pmt.make_dict(),pmt.intern('a'), pmt.PMT_T)
>>> t=pmt.dict_add(pmt.make_dict(),pmt.intern('a'), pmt.PMT_T)
>>> pmt.eqv(s,t)
False
```

I think that `pmt.eqv()` should handle pair type objects.

willcode, 2021-01-06:
How deep should a comparison go, and should order matter?

jacobagilbert, 2021-02-14:
I think requiring order is reasonable and much easier to implement.

jacobagilbert, 2021-02-14:
While its a bit confusing pmt::equal() is the correct function to use on pair-type objects. It does require order, and the first thing it does is test pmt::eqv(). 

```python
>>> pmt.equal(pmt.cons(pmt.PMT_NIL, pmt.PMT_NIL), pmt.cons(pmt.PMT_NIL, pmt.PMT_NIL))
True
>>> s=pmt.dict_add(pmt.make_dict(),pmt.intern('a'), pmt.PMT_T)
>>> t=pmt.dict_add(pmt.make_dict(),pmt.intern('a'), pmt.PMT_T)
>>> pmt.equal(s,t)
True
```

Closed.

closed 2021-02-14 by jacobagilbert