gnuradio/gnuradio #4048: ADI Pluto

herbechun, 2021-01-06:
1/6/2021
I was updating my ADI Pluto's FW and after it updated, at least I think it updated, W10 was not able to see the Pluto as a mass storage device.  W10's device manager does see the USB device driver, but the mass storage device is no longer functioning.  If mass storage is no longer functioning, how can I reset Pluto to factory conditions?
Thank you
-Pluto User

balister, 2021-01-07:
GNU Radio does not provide Pluto hardware support. The ADI wiki suggests posting on the Engineer Zone forum for help. https://wiki.analog.com/university/tools/pluto

closed 2021-01-07 by balister

tfcollins, 2021-01-07:
Please post your question here and I'll help you out for Pluto: https://ez.analog.com/adieducation/university-program/

-Travis

herbechun, 2021-01-07:
Hi Travis

this is great, but I figured it out.

Good to know that ADI has support for Pluto.

By the way, where are you located, are you in North Carolina?

-Herbe

tfcollins, 2021-01-08:
I'm based in LA currently, but the support team for Pluto is spread all over.

-Travis

herbechun, 2021-01-18:
Hi Dr Collins,

thanks for the response, I watched a number of your youtube vid's, all good
stuff!

By the way, what I am trying to do is detect for Iridium burst at 1.626GHz
using Pluto using
the AD9363.  Since the stock antenna isn't tuned for Iridium, I was
thinking of using a Nooelec Iridium
Bundle that has a RHCP Iridium patch antenna + SAWBird (SAW + LAN) and DC
block.  Would you be able to comment on the block diagram before I commit
the everything to BOM?

Herbe Chun
Satelles

tfcollins, 2021-01-19:
Will probably work ok. It's kinda strange they don't have datasheets, but for the price seems to be worth a try.

-Travis

herbechun, 2021-01-19:
Hi Dr. Collins

thank you!

-Herbe

herbechun, 2021-01-22:
Hi Dr Collins

I am now trying to understand more about IIO Scope and configure Pluto and
was
wondering if you could look at the following when you have the chance.

-Herbe Chun, Satelles

herbechun, 2021-01-22:
Hi Dr Collins

sorry had a typo on the attachment, here it is again.

Herbe

tfcollins, 2021-01-22:
@herbechun please open a thread here: https://ez.analog.com/adieducation/university-program

We are getting a bit off-topic here.

-Travis