gnuradio/gnuradio #4050: GNURadio Command Line Shortcut links to a .exe, when it should link to a .bat

ijustlovemath, 2021-01-07:
Shortcut installs with:

    Target:"C:\Program Files\GNURadio-3.8\bin\gnuradio-env.exe"

Should be:

    Target:"C:\Program Files\GNURadio-3.8\bin\gnuradio-env.bat"

I tried to find where this shortcut gets created but couldn't seem to find any code with "gnuradio-env" in it, across almost all of GitHub. Happy to PR if someone points me in the right direction!

marcusmueller, 2021-01-07:
I'm not aware of which source of GNU Radio you're using? I don't release windows installers, so this might not be the optimal way of reporting this (still, good that you reported it; we can figure out together how to get this fixed).

ijustlovemath, 2021-01-07:
I'll edit the title to clarify I'm referring to the desktop shortcut installed by the installer. The installer I used came from here: https://www.castle.cloud/_gnuradio/gnuradio_3.8.2-0_win64_release.exe 

I know this is a minor issue, but for new users to GRC who may rely on desktop shortcuts, it's important to get these minor details right.

ryanvolz, 2021-01-08:
I don't know who has made that installer, but fixing this desktop shortcut is a packaging thing that they would have to handle. As you noted, there's no `gnuradio-env` in this repository anywhere and no built-in desktop shortcuts for Windows. I know because installing a shortcut (albeit for the Start Menu) is something extra I had to do for the [conda packages](https://wiki.gnuradio.org/index.php/CondaInstall) that I maintain.

nickoe, 2021-01-10:
It looks like this should be reported to castle.cloud instead and therefore not an upstream issue.

closed 2021-01-10 by nickoe