gnuradio/gnuradio #4051: cannot import name 'fftpack' from 'numpy.fft'

ChristopheSeg, 2021-01-07:
Hi 
I get this error message when running python scripts containing these two lines of code 
`from numpy.fft import fftpack`
`fftpack.fft(...) `
Only 3 files import fftpack: 
- gr-digital/examples/example_timing.py
- gr-utils/plot_tools/plot_fft_base.py
- gr-dtv/examples/atsc_ctrlport_monitor.py:

This issue is close to [#3259](https://github.com/gnuradio/gnuradio/pull/3259). The fix is quite simple: 

- remove fftpack import  `from numpy.fft import fftpack`
- modify fft call to : `numpy.fft.fft(...)`

This was tested for 3.8 and 3.9 scripts : example_timing.py and plot_fft_base.py (not for atsc_ctrlport_monitor.py since I don't have Thrift installed)

I think maint3.7 is not concerned by this bug since it uses older python package. 
The issue is present on maint3-8 maint3.9 and master branches. 

My question before making PR: 

- do I have to make PR by to these 3 branches ? 

Regards Christophe

nickoe, 2021-01-07:
@catemarie  Usually it is best to create a pull request for master and check if it seem to be a thing that should be backported. Then please state this in the pull request description as nicely as you have done in this issue.

mbr0wn, 2021-02-15:
@ChristopheSeg can you please start with a PR against master? @nickoe is right that we don't have a good process for backports right now, but let's get the master one in first. We regularly backport stuff to 3.9 so that should not be a problem. Then let's see if backporting to 3.8 is just as easy.

closed 2021-02-18 by mormj