gnuradio/gnuradio #4063: Error

7ue4ssA, 2021-01-10:
When I am executing the flowgraph to listen to a FM radio station I get those errors. It continues to go longer and longer and only stops when I kill the flowgraph again. I wonder what that means. 
Am I right that it has to do with the Soundcard ? Or are there other issues I have to fix?
I am running it on Ubuntu 20.04. With an GeForce RTX 2070 and i7 9th Gen.

[+] Available device: 0:0 <Intel> Intel(R) UHD Graphics Coffee Lake Halo GT2
[+] Selected device: Intel(R) UHD Graphics Coffee Lake Halo GT2
aUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaUaU
>>> Done

duggabe, 2021-01-11:
This may be related to https://github.com/gnuradio/gnuradio/issues/3798
Try creating / modifying the ~/.gnuradio/config.conf file as I describe there.

7ue4ssA, 2021-01-11:
> This may be related to #3798
> Try creating / modifying the ~/.gnuradio/config.conf file as I describe there.

I just tried it and was able to run it without the error. I hope it will not appear again. Thanks for the help.

What actually does it change?

duggabe, 2021-01-11:
argilo's explanation on Oct 12, 2020 gives the reasons. It's a matter of buffers. Also I have found that it depends a lot on the specific computers and OSes. That fix is incorporated in 3.9. I will check on back-porting it to 3.8.

closed 2021-01-11 by duggabe

7ue4ssA, 2021-01-11:
Ah okay, Thanks again for the help.