gnuradio/gnuradio #4067: No module named 'oot' in gnuradio 3.8

Hyeongwook-Seo, 2021-01-11:
Hi!
I use the gnuradio 3.8.
I used 'gr_modtool newmod <module_name>' and 'gr_modtool add <block_name>.'
I wrote <block_name>_impl.cc, <block_name>_impl.h, <block_name>.yml and I have my blocks on gnuradio.
I linked blocks, and click the 'Execute the flow graph' button on gnuradio.
Generating and executing the .py file was started.
But the .py file got the 'No module named '<module_name>'' error.
My file cannot import <module_name>.
Help me please.

nickoe, 2021-01-14:
gr_modtool new mod just generates the "boilerplate" template, so I think you need to replace all instances of the `<module_name>` with your own name, including the brackets.

Hyeongwook-Seo, 2021-01-14:
> gr_modtool new mod just generates the "boilerplate" template, so I think you need to replace all instances of the `<module_name>` with your own name, including the brackets.

Of course, my real module name and block name isn't '<module_name>.'
For example, the module name generated by me is 'doa.'
I enter 'gr_modtool newmod doa' in the terminal.
Thus, I have the folder named 'gr-doa' in my PC.

marcusmueller, 2021-01-14:
There's many reasons why your module or blockname can't be imported; none of them are GNU Radio / gr_modtool bugs, luckily!

The most likely reason is that you didn't install your module somewhere your python even looks. 

I'm afraid this is not a GNU Radio bug; closing.

closed 2021-01-14 by marcusmueller