gnuradio/gnuradio #4100: Expose the audio buffer size as options to the audio sink/source blocks

marcusmueller, 2021-01-16:
@dkozel mentions this is only configurable through prefs.

marcusmueller, 2021-01-16:
related #3955

85ayush, 2021-02-23:
I would love to work on this issue, I am new to open source contribution, can you please guide me?

jermainewakeup, 2026-03-05:
Hi, I'm a GSoC 2026 applicant getting familiar with the codebase. I'd like to take a shot at this.

I have experience with audio buffer configuration from embedded audio work on ARM microcontrollers. I'll look at how it currently reads from prefs and surface that as a block parameter. Will update with progress this week.

marcusmueller, 2026-03-05:
@jermainewakeup  Thanks! I'd look at https://github.com/gnuradio/gnuradio/issues/5114 , that's actually nice, because it makes you exercise your C++ skills and allows you to verify you can build the whole GNU Radio tree and run its unit tests, gets you used to our serialization library, and allows you to copy a bit of code from elsewhere. It's also something that allows you to practice your Python when writing the matching unit test, without totally exploding in scope.

closed 2026-03-05 by marcusmueller