gnuradio/gnuradio #4109: GRC doesn´t show variable names only values

reald, 2021-01-17:
At the moment gnuradio-companion shows only values for block attributes in a flow graph but not the variables or expressions behind. This makes it hard to understand and explain more complex flowgraphs because you have to look inside every single block if other variables are used here. So instead of 

**Signal source**
- Sample Rate: 32k
- Frequency: 1k

it should be possible to show block properties on the gnuradio canvas with their expressions behind like:

**Signal source**
- Sample Rate: samp_rate
- Frequency: freq

Additionally it should be possible to show expression and the current value like:

**Signal source**
- Sample Rate: samp_rate (32k)
- Frequency: freq (1k)

The new behaviour should be selectable in the menu.

dkozel, 2021-01-30:
I can see the reasoning behind this and think it would be a nice option to have. When longer values are supplied it won't be very readable, but that's already true of values like vectors or long strings.

jasonuher, 2021-02-24:
> I can see the reasoning behind this and think it would be a nice option to have. When longer values are supplied it won't be very readable, but that's already true of values like vectors or long strings.

I used the same truncate function on the variable names as was used on the values, so the behavior should be the same.

ChristopheSeguinot, 2021-03-01:
It seems that this small issue is closely related to this feature request (from the code change point of view)
Suppose I've a block with Sample Rate defined by variable samp_rate fixed at 1000 Hz. 

- When the block is enabled, it actually shows as "Sample Rate : 1000 Hz"
- When a block is disabled, it's displayed in red as  "Sample Rate : samp_rate"

1. The text should be the same (with/wo string "samp_rate") in both case
2. I think there is no reason to show this in red for disabled blocks.

mormj, 2021-04-28:
Feature merged as part of #4279.  Closing

closed 2021-04-28 by mormj